supermerill/SuperSlicer #2305: Superslicer shape container not a valid drop zone under wayland

mrvanes, 2022-02-02:
### What happened?

This week, I switched my KDE plasma desktop session to wayland. It is quite mature and most of my daily necessities work very well.
Superslicers works like a charm, except that the Shape container area does not like to receive files dragged from dolphin.

I appreciate that wayland/plasma is under heavy development and certainly not ready for everybody, I just wanted to give a heads-up to the devs about the drop issues as it's certainly not a showstopper and I don't know whether this works on a Gnome wayland session e.g.

### Project file & How to reproduce

Start KDE plasma wayland session, start Superslicer and try to drop a file from dolphin (the file explorer) to Superslicer.

### Version

2.3.57.9

### Operating system

Ubuntu 21.10

supermerill, 2022-02-02:
Slic3r and derivatives use [wxWidget](https://www.wxwidgets.org/), so it's not really in our hands.

mrvanes, 2022-02-02:
Tried both AppImages and the tar build, none of them accepts drag-n-drop. Thx for letting me know. As I said, I just wanted to let you know.