supermerill/SuperSlicer #2306: [BR] Weird external lines in surface since 2.3.57.7 to 2.3.57.9 not happend on 2.3.57.3

ldiegos, 2022-02-02:
### What happened?

When slice the surface of the model has inconsistencies in the layers, in particular in the external perimeter when the external perimeter is below or above a champfer or fillet, also if there is any other hole.

Those lines are printed in the models, so, they mess up the final printed part.

SuperSlicer_2.3.57.7 to SuperSlicer_2.3.57.9:
[image]

Printed
[image]

SuperSlicer_2.3.57.3:
[image]

### Project file & How to reproduce

[SuperSlicer_2.3.57.9.zip](https://github.com/supermerill/SuperSlicer/files/7989782/SuperSlicer_2.3.57.9.zip)
[SuperSlicer_2.3.57.3.zip](https://github.com/supermerill/SuperSlicer/files/7989781/SuperSlicer_2.3.57.3.zip)

### Version

2.3.57.3; 2.3.57.7-9;

### Operating system

Windows 10

### Printer model

Two Geetech A10 with a ED3V6

ldiegos, 2022-02-02:
Sorry, i found the issue, is the Filtering resolution. If I try to reduce que model gcode size and free a little my printer resources, going to the 0.05 those lines appeared.

If i reduce the Filtering Resolution to 0.02 they are gone.

Its a little bit odd, that superslicer change the resolution on a straigth line...

0.05:
[image]

0.02:
[image]

supermerill, 2022-02-02:
> Its a little bit odd, that superslicer change the resolution on a straigth line...

alrogithm used:
https://en.wikipedia.org/wiki/Ramer%E2%80%93Douglas%E2%80%93Peucker_algorithm

it's not that it changes the straight line, but it's that the point that starts the straight line is removed.

ldiegos, 2022-02-03:
Thank you supermerill, maybe other algorithm that preserves the first and end point of the curves will be worthy, but knowing this i'm happy to deal with it.

This post can be helpful to others or to be in some kind of wiki.

ldiegos, 2022-07-03:
@supermerill you can close this...