supermerill/SuperSlicer #2307: Add color change/pause "+" button is not shown on the ruler

Tekti, 2022-02-02:
### What happened?

I have noticed since version 2.3.57.5 that "+" button is missing on the ruler in the preview mode. Sometimes, after switching GUI layout options it is shown again. 

[image]

### Project file & How to reproduce

Open SuperSlicer and slice any of the files. On the Gcode Preview button is missing

### Version

2.3.57.9

### Operating system

windows 10

### Printer model

Ender 3 Pro

neophyl, 2022-02-02:
Never happens for me.  I'm running windows 10 too.  

When it happens for you , save the project file and post the zipped up 3mf file saved from File>Save project as.

supermerill, 2022-02-02:
unselect "complete_object"

Tekti, 2022-02-03:
Unselecting the option "complete_object" works, thanks. Is there any reason why it is blocked (other than hitting already printed object)? In my opinion it should be active but with some different tooltip or warning mark.

supermerill, 2022-02-03:
you can't add a color change a certain height if you print each object sequentially as you will go by this height multiple times...

Tekti, 2022-02-03:
This button is not only for color change but also for pause and custom gcodes. 

Scenario example:
I am printing one model few times and at each 10 layer I need to stop, insert magnets and then resume. My pause macro already includes maximum heights of the printed objects so there is no collision path. Each element takes 1.5h to print and then around 30minutes of postprocessing. Instead of waiting for whole batch I can post process individual objects in the meantime printing others without restarting the print.

In my opinion it should be visible at least in expert mode

supermerill, 2022-02-03:
The height slider in "complete sequentially" isn't a height slider, as it returns to 0 at each new object, test it yourself.
So it's a bit difficult to makes it work without any problem.
In the 2.4.1, there is in-place gcode edition, so maybe that's what you should use if you have very specific gcode to inject at a specific place.

zeus1200, 2022-04-24:
@supermerill where can I find the `complete_object` option? I'm not able to find it anywhere. Im using superslicer `2.3.57.12` version

zeus1200, 2022-04-24:
found: Print Settings->output options->complete individual objects

mrhasbean, 2023-01-04:
> This button is not only for color change but also for pause and custom gcodes.

Completely agree. I realise this is a carryover from Prusa / Slic3r but the assertion that you go past that layer multiple times isn't correct when referring to the ruler on the preview screen. When you're using that ruler with complete_objects selected, the "layer" being shown is for the particular object, so anything added at that "layer" would logically only apply for that object. The Gcode is generated sequentially for the individual objects so there should be no impediment to adding anything in there that would normally be possible. 

Functionally I can understand why filament / colour change would be discouraged, but even then, Klipper users or those rolling their own firmware could easily create a macro / custom version of M600 to safely move the carriage away, facilitate the filament change and restore the previous state, based on how they populate the plate when using the complete_objects option.

I agree is should be visible in Expert mode with the caveat that it is "use are your own risk" because no checks can be done to determine if the entered code will wreak havoc on the print.