supermerill/SuperSlicer #2314: Z offset calibration

jeffbearer, 2022-02-04:
**Is your feature request related to a problem? Please describe.**
It would be a real timesaver that if before the bed leveling calibration there was one to determine the appropriate z offset.  

**Describe the solution you'd like**
Imagine a calibration that creates several tabs like the bed level ones but each one is set to a slightly different height simulating z offset.   This would allow you to run one calibration and have 5 or so tabs with some small z offset change.

**Describe how it would work**
Calibration menu like the others.  generate buttons with useful presets,  perhaps one coarse and one fine.

**Describe alternatives you've considered**
I've considered that this can't be generalized to work in most cases.  and it might cause out of of bounds violations in firmwares etc.  idk what do you think?

**Additional context**

supermerill, 2022-02-04:
I think it can scratch/destroy the bed easily, or be useless as they are all too far away.

jeffbearer, 2022-02-04:
Gotcha,  in my case the paper test is too close. and I need to back off.  And I've found using 1 of the tabs and adjusting the z offset in klipper lets me dial it in really well.  but for example today it took 8 tries to dial it in after I did some work on the printer.     So maybe it could be after the bed level test.  ie .   If your prints look like any of the too close.  try this z offset test which will try 5 tabs with increasingly raised z that you can set the z offset?

supermerill, 2022-02-04:
I can add a button to generate a test with only one path at the center, for printer that can only be adjusted via z-offset?