supermerill/SuperSlicer #2318: Time Remaining counts UP on Prusa i3 Mk3S

imhavoc-classic, 2022-02-05:
### What happened?

The "time remaining" indicator on the control panel starts at 0:00 and counts up as the print progresses.

### Project file & How to reproduce

Start a standard print.

### Version

2.3.57.9

### Operating system

Ubuntu 20.04.3 LTS

### Printer model

Prusa i3 Mk3S

supermerill, 2022-02-10:
Can you post a project file where that happens?

imhavoc-classic, 2022-02-12:
I had not seen this happen for a few days, then it happened with this model today.

[jobbery-troubleshooting.zip](https://github.com/supermerill/SuperSlicer/files/8052410/jobbery-troubleshooting.zip)

supermerill, 2022-02-12:
You haven't activated "print remaining times"... I can't see any M73 in the gcode.
Can't do anything.

closed 2022-03-11 by supermerill