supermerill/SuperSlicer #2319: One More Parameter to Determine Segment Length

CCS86, 2022-02-05:
SS has come a long way in giving the user better control of gcode segmentation and making sure that all types of moves honor the settings.

I think we are still missing one more parameter to help govern this: **maximum path deviation**

The current behavior appears to reduce segment length until it hits either the **minimum extrusion length**, or the **maximum G1 per second**. This results in code that runs perfectly fine, but that is overly dense (large files) without benefit.

I think that **maximum G1 per second** should be treated as a "limiter", rather than a "control" to be moved around.

Let's say I set my **minimum extrusion length** to 0.25mm, so that when I print a tiny 1.5mm circle, I get an acceptable segmentation deviation:

[image]

Then, when I print a 60mm circle, I am getting similar segment lengths, but now my segmentation deviation is near zero:

[image]

That segment length could be 4.7 times longer with a matching path deviation:

[image]

Adding a prescribed **maximum path deviation** value would set a practical limit to manage the size of gcode files, letting **minimum extrusion length** & **maximum G1 per second** act the limiters they are meant to be.

Thoughts?

CCS86, 2022-02-05:
I have tried using:

 **filtering > resolution**, but that makes even the small arc very coarse.

 **filtering > internal resolution** doesn't seem to change segment length / count

But I think both of these are meant to change the slicing itself, not the decomposition into gcode segments.

CCS86, 2022-02-25:
Any thoughts @supermerill ?

CCS86, 2022-03-07:
I would tend to think it would go in the printer profile, along with max G1/sec.

supermerill, 2022-03-07:
I can make it a float or % of the print resolution.

CCS86, 2022-03-07:
I would definitely vote for float, for more direct control. 

Would the value represent a max deviation, like I sketched above?

supermerill, 2022-03-07:
> Would the value represent a max deviation, like I sketched above?

See the Wikipedia article (there is an animation showing how it remove points). I think it's what you sketch.

CCS86, 2022-07-30:
Hey @supermerill I just wanted to check in on this one. You labeled it as easy, and it would be a real help in tuning print quality and file size.

CCS86, 2022-09-27:
@supermerill any progress?