supermerill/SuperSlicer #2321: Add "Z-Offset" filament override

Arakon, 2022-02-06:
**Is your feature request related to a problem? Please describe.**
Different materials like different Z-Offsets.. i.e. PETG is generally printed with slightly less squish, while PLA likes a bit more.

**Describe the solution you'd like**
A filament setting override that allows to change Z-Offset specific to a filament, to avoid having to change it in the printer setting or having 2 printer profiles for different materials.

ManuelMcLure, 2022-02-06:
The workaround I use for this is code like this in my filament-specific start gcode for PETG:
```
G0 Z2
G92 Z1.94 ; For an equivalent 0.06mm raise
```

sprior, 2022-02-11:
I agree with this request - some materials (like the mentioned PETG) like less first layer squish than others.

supermerill, 2022-02-11:
duplicate of #171 and #602

t0m-r1dd1e, 2022-05-20:
I'd also love to see this implemented. My use case is that I want more z offset for flexibles.

user897943, 2022-09-08:
Just googled this and was thinking exactly that - "why isn't this on the filament overrides?"
Seems a bit of wasted space to have an overrides section with just retraction settings... Started using SuperSlicer yesterday. Hopefully see this implemented soon, it's been open since February 2022...

zeroemso, 2022-09-15:
i would like to see that too =)

matt-rendall, 2023-01-04:
FWIW I'm working around this by passing {filament_type} into my print_start macro where I have adjustment values set per filament type, which I then use with SET_GCODE_OFFSET Z_ADJUST at the very end of my start sequence.

pimmen85, 2023-03-13:
> The workaround I use for this is code like this in my filament-specific start gcode for PETG:
> 
> ```
> G0 Z2
> G92 Z1.94 ; For an equivalent 0.06mm raise
> ```

this is what i have been looking for, just as a HINT, if i have good z-offset right now with PETG, how do i go about making that specific to the Filament G-code?

and vice versa, with a tighter z-offset for PLA?

ipatch, 2023-03-13:
> this is what i have been looking for, just as a HINT, if i have good z-offset right now with PETG, how do i go about making that specific to the Filament G-code?
>
> and vice versa, with a tighter [...]

add it to the custom gcode for the specific filament you have setup in superslicer

[image]

BrockPlaysFortnite, 2023-03-17:
> add it to the custom gcode for the specific filament you have setup in superslicer
>
> [image]

Thank you I was also searching for a way of custom Z for PETG