supermerill/SuperSlicer #2323: SuperSlicer crashes on startup

helfrichmichael, 2022-02-06:
### What happened?

**Behavior Experiened:**

I am consistently unable to launch SuperSlicer without it crashing inside of an active Docker container I'm developing (launches for a few seconds and then crashes).

**Expected Behavior:**

When I launch SuperSlicer, it should stay open without crashing. This Docker container works with PrusaSlicer, so it's kind of bizarre that it's crashing given they should have a similar dependency stack.

**Version:**

2.3.57.10 via https://github.com/supermerill/SuperSlicer/releases/download/2.3.57.10/SuperSlicer_2.3.57.10_linux64_220203.tar.zip

**Logs:**

```
slic3r@b0fea54f8ce5:/slic3r/slic3r-dist$ ./superslicer --loglevel 4
[2022-02-06 17:17:38.748696] [0x00007f4ca2c265c0] [debug]   full path: 1386180253825838897
[2022-02-06 17:17:38.748722] [0x00007f4ca2c265c0] [debug]   single instance: undefined. other params: /slic3r/slic3r-dist/bin/superslicer;--loglevel;4
[2022-02-06 17:17:38.748725] [0x00007f4ca2c265c0] [debug]   full lock path: /home/slic3r/.config/SuperSlicer/cache/1386180253825838897.lock
[2022-02-06 17:17:38.748746] [0x00007f4ca2c265c0] [info]    instance check: Another instance not found or single-instance not set.
[2022-02-06 17:17:38.784112] [0x00007f4ca2c265c0] [info]    Checking if indices need to be installed from resources...

(superslicer:328): GLib-GObject-WARNING **: 17:17:38.885: invalid (NULL) pointer instance

(superslicer:328): GLib-GObject-CRITICAL **: 17:17:38.885: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:328): GLib-GObject-WARNING **: 17:17:38.885: invalid (NULL) pointer instance

(superslicer:328): GLib-GObject-CRITICAL **: 17:17:38.885: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:328): GLib-GObject-WARNING **: 17:17:38.885: invalid (NULL) pointer instance

(superslicer:328): GLib-GObject-CRITICAL **: 17:17:38.885: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:328): GLib-GObject-WARNING **: 17:17:38.885: invalid (NULL) pointer instance

(superslicer:328): GLib-GObject-CRITICAL **: 17:17:38.885: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
[2022-02-06 17:17:38.948285] [0x00007f4ca2c265c0] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 98MB; Shared memory: 65MB; Private memory: 33MB; Peak memory usage: 98MB
[2022-02-06 17:17:38.949406] [0x00007f4c55ffb700] [error]   DBus Connection Error: /usr/bin/dbus-launch terminated abnormally without any error message
[2022-02-06 17:17:38.949422] [0x00007f4c55ffb700] [error]   Dbus Messages listening terminating.

(superslicer:328): GLib-GObject-WARNING **: 17:17:38.994: invalid (NULL) pointer instance

(superslicer:328): GLib-GObject-CRITICAL **: 17:17:38.994: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:328): GLib-GObject-WARNING **: 17:17:38.994: invalid (NULL) pointer instance

(superslicer:328): GLib-GObject-CRITICAL **: 17:17:38.994: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:328): GLib-GObject-WARNING **: 17:17:38.994: invalid (NULL) pointer instance

(superslicer:328): GLib-GObject-CRITICAL **: 17:17:38.994: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:328): GLib-GObject-WARNING **: 17:17:38.994: invalid (NULL) pointer instance

(superslicer:328): GLib-GObject-CRITICAL **: 17:17:38.994: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:328): GLib-GObject-WARNING **: 17:17:38.994: invalid (NULL) pointer instance

(superslicer:328): GLib-GObject-CRITICAL **: 17:17:38.994: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:328): GLib-GObject-WARNING **: 17:17:38.994: invalid (NULL) pointer instance

(superslicer:328): GLib-GObject-CRITICAL **: 17:17:38.994: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:328): GLib-GObject-WARNING **: 17:17:38.994: invalid (NULL) pointer instance

(superslicer:328): GLib-GObject-CRITICAL **: 17:17:38.994: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
[... 128 lines cut ...]

(superslicer:328): Gtk-CRITICAL **: 17:17:39.024: gtk_widget_set_size_request: assertion 'height >= -1' failed

(superslicer:328): Gtk-CRITICAL **: 17:17:39.024: gtk_widget_set_size_request: assertion 'height >= -1' failed
[2022-02-06 17:17:39.049200] [0x00007f4ca2c265c0] [debug]   TriangleMesh::repair() started
[2022-02-06 17:17:39.049451] [0x00007f4ca2c265c0] [debug]   TriangleMesh::repair() finished
[2022-02-06 17:17:39.049500] [0x00007f4ca2c265c0] [debug]   TriangleMesh::repair() started
[2022-02-06 17:17:39.049632] [0x00007f4ca2c265c0] [debug]   TriangleMesh::repair() finished
Segmentation fault
```

### Project file & How to reproduce

Here is the Docker container I'm building with it's dependencies https://github.com/helfrichmichael/superslicer-novnc

Using Docker and this repo, you can reproduce with the following command:
`docker build -t superslicer-novnc . && docker run --detach --volume=superslicer-novnc-data:/configs/ -p 8080:8080 -e SSL_CERT_FILE="/etc/ssl/certs/ca-certificates.crt" --name=superslicer-novnc superslicer-novnc`

### Version

2.3.57.10 

### Operating system

Debian 10

helfrichmichael, 2022-02-08:
I ran it with loglevel 5 this time and that's a bit more telling, but nothing stands out. Seems to consistently crash in the same spot in this. `strace` wasn't of much use from what I could gather with the output.

I'm still looking into it on my end, but I haven't come across a fix and I've tried this across multiple Debian versions (Bullseye in the run below) and also Ubuntu versions (including 18.04). As well as the gtk2 image with the `--extract-and-run` option

```
slic3r@a44716e919de:/slic3r/slic3r-dist$ ./superslicer --loglevel 5
[2022-02-08 01:40:25.129727] [0x00007fc01d9ea5c0] [debug]   full path: 1386180253825838897
[2022-02-08 01:40:25.129751] [0x00007fc01d9ea5c0] [debug]   single instance: undefined. other params: /slic3r/slic3r-dist/bin/superslicer;--loglevel;5
[2022-02-08 01:40:25.129754] [0x00007fc01d9ea5c0] [debug]   full lock path: /home/slic3r/.config/SuperSlicer/cache/1386180253825838897.lock
[2022-02-08 01:40:25.129778] [0x00007fc01d9ea5c0] [info]    instance check: Another instance not found or single-instance not set.
[2022-02-08 01:40:25.173525] [0x00007fc01d9ea5c0] [info]    Checking if indices need to be installed from resources...
[2022-02-08 01:40:25.174823] [0x00007fc01d9ea5c0] [trace]   System language detected (user locales and such): en_US
[2022-02-08 01:40:25.175235] [0x00007fc01d9ea5c0] [trace]   Best translation language detected (may be different from user locales): en_US
[2022-02-08 01:40:25.175259] [0x00007fc01d9ea5c0] [trace]   Switching wxLocales to en_US

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.277: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.277: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.277: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.277: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.277: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.277: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.277: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.277: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
[2022-02-08 01:40:25.348229] [0x00007fc01d9ea5c0] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 117MB; Shared memory: 73MB; Private memory: 43MB; Peak memory usage: 117MB
[2022-02-08 01:40:25.350037] [0x00007fbf71ffb700] [error]   DBus Connection Error: /usr/bin/dbus-launch terminated abnormally without any error message
[2022-02-08 01:40:25.350051] [0x00007fbf71ffb700] [error]   Dbus Messages listening terminating.

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.396: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.396: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.396: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.396: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.396: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.396: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.396: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.396: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.396: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.396: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.396: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.396: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.397: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.397: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.397: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.397: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:1774): GLib-GObject-WARNING **: 01:40:25.397: invalid (NULL) pointer instance

(superslicer:1774): GLib-GObject-CRITICAL **: 01:40:25.397: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

[... 173 lines cut ...]
< X-GitHub-Media-Type: github.v3; format=json
< Link: <https://api.github.com/repositories/116708603/releases?page=2>; rel="next", <https://api.github.com/repositories/116708603/releases?page=3>; rel="last"
[... cut ...]
< X-RateLimit-Limit: 60
< X-RateLimit-Remaining: 59
< X-RateLimit-Reset: 1644288025
< X-RateLimit-Resource: core
< X-RateLimit-Used: 1
< Accept-Ranges: bytes
< Transfer-Encoding: chunked
< X-GitHub-Request-Id: C689:3A81:56139C:D1F461:6201CA09
< 
* Connection #0 to host api.github.com left intact
[2022-02-08 01:40:26.089489] [0x00007fbf5401c700] [info]    Syncing configuration cache
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Aborted
```

neophyl, 2022-02-08:
Which version of Prusa Slicer does it work with ?  There were LOTS of issues with crashing in previous versions of PS , many throughout the 2.4 alphas and betas etc.  Even now with 2.4.1 there are still some.  Given that SuSi is currently built on the 2.3 PS releases then the dependencies might be more different then expected at the moment.

helfrichmichael, 2022-02-08:
> Which version of Prusa Slicer does it work with ? There were LOTS of issues with crashing in previous versions of PS , many throughout the 2.4 alphas and betas etc. Even now with 2.4.1 there are [...]

For PrusaSlicer, I've tested 2.4.0 and the 2.4.1 betas and those worked without issue. I can definitely try this evening with older 2.3 releases if that might be more telling.

supermerill, 2022-02-09:
Can't replicate on my ubuntu 20.04

can you try with https://github.com/supermerill/SuperSlicer/actions/runs/1814764921 ? they are using the debug build, so they should be able to output a good stacktrace.

helfrichmichael, 2022-02-10:
> 

Thanks @supermerill for the recommendation. Using the generated appimage file and the following dependencies I am getting the same behavior mentioned earlier. I'm also not seeing anything telling in these logs for the debug build vs GitHub release.

Dependencies:
```
FROM jlesage/baseimage-gui:ubuntu-20.04
RUN add-pkg xterm sudo sed wget curl fuse libgtk2.0-dev libglew-dev libudev-dev libdbus-1-dev locales locales-all

Running the AppImage with --loglevel 5
```

Logs from Docker Inspect:
```
04:05:40: Debug: Adding duplicate image handler for 'JPEG file'

(superslicer:942): Gtk-CRITICAL **: 04:05:40.882: IA__gtk_window_resize: assertion 'width > 0' failed

(superslicer:942): GLib-GObject-WARNING **: 04:05:40.910: invalid (NULL) pointer instance

(superslicer:942): GLib-GObject-CRITICAL **: 04:05:40.910: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:942): GLib-GObject-WARNING **: 04:05:40.910: invalid (NULL) pointer instance

(superslicer:942): GLib-GObject-CRITICAL **: 04:05:40.910: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:942): GLib-GObject-WARNING **: 04:05:40.910: invalid (NULL) pointer instance

(superslicer:942): GLib-GObject-CRITICAL **: 04:05:40.910: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:942): GLib-GObject-WARNING **: 04:05:40.910: invalid (NULL) pointer instance

(superslicer:942): GLib-GObject-CRITICAL **: 04:05:40.910: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

10/02/2022 04:05:40 client 1 network rate 8526.6 KB/sec (164268.6 eff KB/sec)

10/02/2022 04:05:40 client 1 latency:  0.7 ms

10/02/2022 04:05:40 dt1: 0.0048, dt2: 0.0105 dt3: 0.0007 bytes: 128018

10/02/2022 04:05:40 link_rate: LR_LAN - 1 ms, 8526 KB/s

04:05:40 AM: Debug: Unrecognized accel key 'Dèl', accel string ignored.

04:05:41 AM: Debug: "Shift+Tab" is not supported as a keyboard accelerator with GTK

(superslicer:942): GLib-GObject-WARNING **: 04:05:41.018: invalid (NULL) pointer instance

(superslicer:942): GLib-GObject-CRITICAL **: 04:05:41.018: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:942): GLib-GObject-WARNING **: 04:05:41.018: invalid (NULL) pointer instance

(superslicer:942): GLib-GObject-CRITICAL **: 04:05:41.018: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:942): GLib-GObject-WARNING **: 04:05:41.018: invalid (NULL) pointer instance

(superslicer:942): GLib-GObject-CRITICAL **: 04:05:41.018: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:942): GLib-GObject-WARNING **: 04:05:41.018: invalid (NULL) pointer instance

(superslicer:942): GLib-GObject-CRITICAL **: 04:05:41.018: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:942): GLib-GObject-WARNING **: 04:05:41.018: invalid (NULL) pointer instance

(superslicer:942): GLib-GObject-CRITICAL **: 04:05:41.018: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(superslicer:942): GLib-GObject-WARNING **: 04:05:41.018: invalid (NULL) pointer instance

(superslicer:942): GLib-GObject-CRITICAL **: 04:05:41.018: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

[... 477 lines cut ...]

[2022-02-10 04:29:05.536558] [0x00007f8e17fb05c0] [info]    Checking for cached configuration updates...

[2022-02-10 04:29:05.536587] [0x00007f8e17fb05c0] [info]    Confing bundle not installed for vendor FLSun, skipping: 

[2022-02-10 04:29:05.536594] [0x00007f8e17fb05c0] [info]    Confing bundle not installed for vendor Anycubic, skipping: 

[2022-02-10 04:29:05.536597] [0x00007f8e17fb05c0] [info]    Confing bundle not installed for vendor LulzBot, skipping: 

[2022-02-10 04:29:05.536605] [0x00007f8e17fb05c0] [info]    Confing bundle not installed for vendor BIBO, skipping: 

[2022-02-10 04:29:05.536636] [0x00007f8e17fb05c0] [info]    Confing bundle not installed for vendor PrusaResearch, skipping: 

[2022-02-10 04:29:05.536643] [0x00007f8e17fb05c0] [info]    Confing bundle not installed for vendor TriLAB, skipping: 

[2022-02-10 04:29:05.536651] [0x00007f8e17fb05c0] [info]    Confing bundle not installed for vendor Creality, skipping: 

[2022-02-10 04:29:05.536652] [0x00007f8e17fb05c0] [info]    No configuration updates available.

[2022-02-10 04:29:05.548454] [0x00007f8d6ddfc700] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`

[2022-02-10 04:29:06.040296] [0x00007f8d6ddfc700] [terminate called after throwing an instance of 'std::logic_error'

  what():  basic_string::_S_construct null not valid
```

helfrichmichael, 2022-02-11:
I tested the following versions: `2.3.0, 2.3.3, 2.4.0-alpha-1, and 2.4.0-alpha2`.

All of the versions except 2.4.0-alpha2 crash. I then looked through the changelog of 2.4.0-alpha2 and I stumbled on https://github.com/prusa3d/PrusaSlicer/issues/3969 which was for `Linux specific: Fixed crash when $USER environment variable was not set`. 

I went in and that's my issue to a tee.

I'm going to make a Pull Request shortly and build it on my end to validate things if that's okay with you all. Basically it all boils down to Docker not setting the $USER variable.

supermerill, 2022-02-11:
If the current 2.4 nightly of superslicer is also working, don't bother.

helfrichmichael, 2022-02-11:
Thanks, using https://github.com/supermerill/SuperSlicer/actions/runs/1817605679 I was able to successfully run this in various Docker instances using the GUI. 

Thanks for the quick response/support!

closed 2022-04-01 by supermerill