supermerill/SuperSlicer #2326: Turn fan on/up when printing supports

lousassole, 2022-02-07:
**Is your feature request related to a problem? Please describe.**
While printing PETG or ABS, supports tend to bond to the part as I generally don't print with the cooling fan on for better layer adhesion.

**Describe the solution you'd like**
The ability to have the fan turn on while printing supports (above the first couple layers) either at the top of the support, or the entire time supports are being printed. This would make layer adhesion on supports worse, making removing supports much easier.

**Describe how it would work**
Add an option for turning on part cooling fans while printing supports. The ability to configure fan speed would be nice, but not necessary for most I'd imagine, but would be appreciated for those that do need it.

**Describe alternatives you've considered**
Manually adding gcode to turn fan on when printing supports, then manually adding gcode to turn fan off when finished; or doing it by hand on the printer's menu during a print.

**Additional context**
PETG and ABS would print more nicely and have a better time with support removal if fans were on while printing supports.

supermerill, 2022-02-08:
Currently, you can activate it manually via feature_gcode field.
you have to put in it something like `{if extrusion_role == "SupportMaterial"}M106 S255{else}M106 S{min_fan_speed}{endif}`

lousassole, 2022-02-09:
awesome, thanks! i figured there was an easy way to add this via custom gcode scripts, but had no idea how (not super familiar with gcode yet).

levleon13, 2022-02-10:
so this will not be added as a menu toggle?
super helful function, how petg users arent seeing this as supports salvation ?!

supermerill, 2022-02-10:
yes.
Add also the `and layer_z >3` to the second if so it won't erase fan when it's not needed (as it can erase the higher fan from low layer time)

lousassole, 2022-02-11:
awesome, thank you!

legend069, 2022-02-11:
[Body7_testing.zip](https://github.com/supermerill/SuperSlicer/files/8046963/Body7_testing.zip)

below is the one i will use, but during testing, I've noticed a bug perhaps ?
with `disable_fan_first_layers`
if you change that value from 1 to 3
the support material at layer 18 doesn't not allow the extrusion role change and with the `disable_fan_first_layers`
but setting `disable_fan_first_layers` to 1 or 2 doesn't allow the extrusion role change to run.

i am still on 2.3.57, so not sure if this has been fixed or not.

`{if extrusion_role == "SupportMaterialInterface" and layer_z > disable_fan_first_layers} M106 S255  {elsif last_extrusion_role == "SupportMaterialInterface" and layer_z > disable_fan_first_layers } M106 S{min_fan_speed}{endif}`

supermerill, 2022-02-11:
That's because you used layer_z and not layer_num, no?

If you want to start at the 4th layer, use  `layer_num>3`
If you want to start after 3mm height, use  `layer_z>3`

legend069, 2022-02-11:
yes that did it. but looking into it again.
it looks like having disable_fan_first_layers = 3 or 12

is applying that value for the support area and the model area independently 
so whats happening is (using disable_fan_first_layers  = 12 in this case)
it does the 12 layer of support
this is also the 12th layer of the model
and then at layer 24 it applies the fan

[Body7_testing.zip](https://github.com/supermerill/SuperSlicer/files/8047894/Body7_testing.zip)

i would expect it to apply disable_fan_first_layers at the total layer number of 12, but it does it at 24

after adjusting support_material_contact_distance_top it behaves correctly.
adding and removing the "between extrusion role change" also changes it a bit.

supermerill, 2022-02-11:
I forgot, `min_size_pseed` is a 0-100 value, you ahve to use
`M106 S{if fan_percentage}{min_fan_speed}{else}{(min_fan_speed*255)/100}{endif}`

levleon13, 2022-02-14:
Hi guys, i have copied the 
`{if extrusion_role == "SupportMaterialInterface" and layer_z > disable_fan_first_layers} M106 S255 {elsif last_extrusion_role == "SupportMaterialInterface" and layer_z > disable_fan_first_layers } M106 S{min_fan_speed}{endif}`

and i have similar issue with 1st 3 layers that are also cooled by the fan. My coding knowledge is limited, i did see that [legend069](https://github.com/legend069) has the same issue, but i did not understand what should i copy to get rid of first 3 layers of cooling.

levleon13, 2022-02-21:
> 

did exactly that. still get 100% on support at 1st layer.

[image]
[image]

legend069, 2022-02-22:
> > 
> 
> did exactly that. still get 100% on support at 1st layer.
> 
> [image] [image]

you still have the first line that is the problem. just copy/paste what i have.

below is what i have and it seems like it works fine.

```
{if extrusion_role == "SupportMaterialInterface" and layer_num > disable_fan_first_layers} M106 S{(bridge_fan_speed*255)/100}  {elsif last_extrusion_role == "SupportMaterialInterface" and layer_num > disable_fan_first_layers } M106 S{(min_fan_speed*255)/100} {endif}

{if extrusion_role == "SupportMaterial" and layer_num > disable_fan_first_layers}  M106 S{(external_perimeter_fan_speed*255)/100}{elsif last_extrusion_role == "SupportMaterial" and layer_num > disable_fan_first_layers } M106 S{(min_fan_speed*255)/100} {endif}
```

@supermerill  it looks like "fan_below_layer_time" is prioritised over "extrusion role change g-code" should this be correct?

supermerill, 2022-02-22:
@legend069

Yes, all of the things that depend on time is done as a post-process on the gcode, as you need the gcode to do the time simulation.

levleon13, 2022-02-24:
> you still have the first line that is the problem. just copy/paste what i have.
>
> below is what i have and it seems like it works fine.
>
> ```
> {if extrusion_role == "SupportMaterialInterface" and [...]

thanks a lot !

levleon13, 2022-02-24:
now i get this on my mainsailos:

Error on 'M106 S-2': S must have minimum of 0.0

legend069, 2022-02-24:
> now i get this on my mainsailos:
> 
> Error on 'M106 S-2': S must have minimum of 0.0

i'll suggest looking at your fan cooling settings and remove any negative values ie "-1"

levleon13, 2022-02-27:
> > now i get this on my mainsailos:
> > Error on 'M106 S-2': S must have minimum of 0.0
> 
> i'll suggest looking at your fan cooling settings and remove any negative values ie "-1"

none are at -1, and still same error.

666slc666, 2022-03-07:
I'd love to see this feature too. It's only a matter of days before I slip with the razor blade while cleaning up where the support has stuck to the part.

supermerill, 2022-03-07:
> now i get this on my mainsailos:
> 
> Error on 'M106 S-2': S must have minimum of 0.0

can you post your project file? I don't see how that can happens, as it's protected by a `max ( 0, fan_speed )` Maybe something weird with the wipe tower or a post-porcessing.

LordFlashmeow, 2022-07-07:
@legend069 the script gets confused if the extruder transitions from support material interface to support material (eg. on another object). It sets the fan speed low because the `SupportMaterialInterface` speed is overwritten by `last_extruder_role == "SupportMaterial"`

I restructured the conditionals so that the fan speed is reset only if the current role isn't support or support interface. (And also a material check so it doesn't mess with my PLA prints)

```
{if filament_type == "ABS" and layer_num > disable_fan_first_layers} ; reduce support adhesion by increasing fan
{if extrusion_role == "SupportMaterialInterface"} M106 S{(bridge_fan_speed*255)/100}
{elsif extrusion_role == "SupportMaterial"} M106 S{(external_perimeter_fan_speed*255)/100}
{elseif last_extrusion_role == "SupportMaterial" or last_extrusion_role == "SupportMaterialInterface"}M106 S{(min_fan_speed*255)/100}
{endif}{endif}
```