supermerill/SuperSlicer #2328: Layer Heater

adamrumjahn, 2022-02-07:
Not a problem, just a feature.  We are experimenting with a heater to heat up the layer.  Currently this is a flat ceramic heater, but we're also interested in heating the cooling air to get a precise cooling temperature.  It would be great if this could be added as an additional heater under the "Chamber Heater".  

All we need is just the variable [layer_temperature] to be defined so that it can be called under the custom g-codes.

**Describe alternatives you've considered**
We're currently entering this in manually under Filament Custom G-codes.  It works, but not the cleanest solution.

adamrumjahn, 2022-02-07:
I guess to give some more context, the system that we're creating is enabling better inter-layer bonding by preheating the layer before the extruder nozzle gets to it.  It also makes cooling less dramatic, thus reducing internal stresses in the part.

supermerill, 2022-02-07:
Add it as a custom variable. It's in the "notes" tab.
Be sure to be on the latest release.
Add a "default" before using it, so it won't emit an error if you are using a filament where it's not defined.

adamrumjahn, 2022-02-07:
Thanks for the quick reply.

Sorry this was meant to be a feature request.  

Currently, I'm doing this using Filament Custom G-codes.  Between your suggestion to use custom variables and the custom g-code, they seem to be about the same effort.  But the OCD in me would like to see a Layer Temperature input on the Filament Temperatures tab under the Chamber temperature here:

[image]

supermerill, 2022-02-07:
I may remove the "chamber temperature" as it can be set via custom variable. For what it's used anyway...

adamrumjahn, 2022-02-07:
I'm actually using it at the moment. It's quite useful for me.

Would it be possible to define a custom variable and have it popup in the
Filament temperature tab? Or perhaps an option to add heaters (similar to
Simplify3D)? Maybe this is a simpler approach, to just let users define as
many new heaters as they need.

supermerill, 2022-02-07:
In `resources/ui_layout/filament.ui`, if you move the lines for custom variables (line 97&98) from notes page to the first page, you will be able to modify them from the first page, next to the other temps. It's the most I can offer you for the moment.

In the long run, I want to add "custom widget" to be able to present a simpler interface, or shortcuts for some features, like the ones in the right platter panel. And maybe that will allow you to create custom widget for custom vars, but no promise (won't do it if it's too complicated to implement).