supermerill/SuperSlicer #2329: Perimeter overlap control not actually doing much

CuredPrusa, 2022-02-07:
### Version
Version: 2.3.57.10
Build: SuperSlicer_2.3.57.10

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 34,287MB

### Behavior
* _Describe the problem_
I'd like to use perimeter_overlap treshold in percentage when I select thin_perimeters_all. My model is a (inside conical) cylinder with thin walls, where I'd like to use only walls (6 perimeters max), not infill, for sake of precision and simplicity. When I change perimeter_overlap from 1 to 99, the change is barely noticable on sliced preview. At 1% it is still ovelaping a lot!
* _Steps needed to reproduce the problem_
Make some cylinder in slightly conical shape to generate more perimeters as z rises and you'll see behaviour of perimeter_overlap
* _Expected Results_
When changing perimeter_overlap parameter, perimeters with higher overlap than set percentage are expected to be ingnored.
* _Actual Results_
Regardless of percentage, doubled perimeters are squeezed tight and deformed, because there is no (or not enough) control of overlap percentage 
  * _Screenshots if useful_
[image]

In screenshot is visible how middle perimeters (inner wall) are ovelapping significantly, although overlap is set to 1%

_Is this a new feature request?_
no

supermerill, 2022-02-07:
you have selected "overlapping perimeter". deactivate it.

CuredPrusa, 2022-02-07:
Sorry, I have just realized that "perimeter_overlap" parameter is related to connection (bonding) between perimeter lines and not the actual double extrusion which overlapping perimeter does. I was hoping I would get something similar to "gap fill" with perimeters, but now I see this is not the case. I would use "gap fill" if I could locate it just to one particular section, not the whole print. Therefore I hoped packed perimeters would do the trick. 

[image]

I have just discovered SuperSlicer and I'm extremely grateful for its existance. SuperSlicer saved my nerves, eaten both by Cura and Prusa. SS is very precise, the seam is neat and generated gcode delivers expected consistency. And all those available parameters are really handy to guide the slicer towards best results!

closed 2022-02-09 by CuredPrusa