supermerill/SuperSlicer #2332: Seam painting/Seam position broken

666dsa666, 2022-02-08:
### What happened?

Seam position change are not taken into account.
Using paint-on seam or seam position have little to no effect even when forcing to not use anywhere else on the part.

[image]
[image]
[image]

Using seam sphere : 
[image]

Results are the exact same

I repaired the file using Azure repair tool to no avail but I believe it might be a file issue as seam position changed when repaired but still those options seems to not be taken into account.

Same issue on different part (created using Tinkercad myself): 
[image]
[image]

Changing the option to Aligned, Rear, Random or Cost Based have no effect either.

### Project file & How to reproduce

https://www.thingiverse.com/thing:2908201

Repairint file change seam potition but painting them is still not taken into account.

### Version

2.3.57.9 and .10

### Operating system

Windows 11

### Printer model

Prusa MK3s

666dsa666, 2022-02-08:
Using an alternative file issue is still present : 
https://www.thingiverse.com/thing:2425549/files

[image]
[image]
[image]

supermerill, 2022-02-08:
Can't reproduce
[image]

please send your project file as requested.

666dsa666, 2022-02-08:
[TS_wristrest_1.0_repaired.zip](https://github.com/supermerill/SuperSlicer/files/8026021/TS_wristrest_1.0_repaired.zip)
Thanks for the quick answer ! Here it is. Sorry I did not understand the project file part.

supermerill, 2022-02-08:
Currently, when you are using `perimeter_loop`, it uses another seam algo as it serch for the smallest possible joint.
It has to be improved, and maybe use the new seram tools that are now available (but wasn't when created)