supermerill/SuperSlicer #2345: Superslicer will not run correctly on a Surface Pro

Ender0331, 2022-02-10:
**Bug description**
You are able to install the program and work all the menus to select your printer profile or in my case import the RatRig vcore profile but when you go to import a stl file it dose not show on the build plate and I am unable to rotate the bed as well as the bed graphic glitching out and or changing to a black screen

**Expected behavior**
On my surface pro x superslicer will not run right not showing anything on the bed or even being able to rotate the bed 

!!>> Project File <<!!
NA

**To Reproduce**
Steps to reproduce the behavior:
Open superslicer on my surface pro x

**Screenshots**
NA

**Desktop (please complete the following information):**
 - OS: Windows 11 ARM

**Additional context**
Add any other context about the problem here.

supermerill, 2022-02-10:
Sadly, I can't support windows arm. I don't have one and I don't have the time to debug for it.
If you or someone can debug the app, or even better create a build script for the platform, I'll happily merge them.

Ender0331, 2022-02-10:
Hmm maybe I can get prusa to do the work for you. There slicer acts the same way. If they debug the script for there slicer would you be able to adapt it to yours?

supermerill, 2022-02-10:
maybe, yes.