supermerill/SuperSlicer #2347: last linux version can't save modification , only cancel buton

salocain, 2022-02-10:
after my last post i start superslicer from scratch 

i configure the software and when i want to quit --> he ask for unsaved config but only cancel buton 

only solution --> do a save on each parametre

[image]

tpikalek, 2022-02-14:
This happens also when switching profiles. There is only Transfer and Cancel. Discard and Save buttons are missing.
[image]

tpikalek, 2022-02-15:
Here is where the buttons are disabled if `__WXGTK__` is defined. I suppose only the cancel button should change based on the condition.
https://github.com/supermerill/SuperSlicer/blob/fbc5d5b8076e2072d6f8289921d9f62cfaaee1c8/src/slic3r/GUI/UnsavedChangesDialog.cpp#L871-L885

closed 2022-07-14 by supermerill