supermerill/SuperSlicer #2348: Too much flow in fills and flow control of solid fill is missing

CuredPrusa, 2022-02-10:
### Version
Version 2.3.57
Build:     SuperSlicer_2.3.57.10

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 34,287MB

### Behavior
* _Describe the problem_
All fills (except the infill) are to rich. They are ovelflowing and ruining the layer. Visual animation looks nice, but on printer it's terrible. Walls are OK, infill OK. Then I adjust the flow and get somewhat better result for bottom and top (75%!). 

Parameters: 
Fill patern is not "(filled)", 
"Enforce_full_fill_volume" is off, 
nozzle is 0.8, 
line is 0.82, 
layer is 0.32, 
PLA. 

There's no flow control of solid fill, but I have found a workaround if just one top solid layer is enough -I reduce "over_bridge_flow_ratio". However, that solid layer above the bridge is not changing the flow then (why??), but it is changing a pattern density. Ok, this could actually fix the overflow problem, but it's not right.

* _Steps needed to reproduce the problem_
Change the flow of "above the bridge" and observe

* _Expected Results_
Flow change should just reduce extrusion amount, pattern and spacing should not change. 

* _Actual Results_
Solid layer is overflowing, there's nothing i can do about it. 

[image]
[image]
[image]
[image]
[image]

I think this is a bug, at least related to "above the bridge" flow control.
But here's a new feature request:
To fine tune the perfection of a print, each line type should have its own flow adjustment.

supermerill, 2022-02-10:
project file?

CCS86, 2022-02-11:
I have found the "flow over bridge" to be non-functional.

CuredPrusa, 2022-02-11:
Thanks for response supermerill. 
Unfortunatelly I can't get any flows running good. I expect 100% to be matching by calculation and then I play with more or less to fine tune. But 100% in my case is just bulges.
Lines like perimeters, brim, infill are fine. 
Could it be that calculation for big nozzle has some bug?
Here's the project.
[TestFlow_BE.zip](https://github.com/supermerill/SuperSlicer/files/8050844/TestFlow_BE.zip)

CuredPrusa, 2022-02-11:
> I have found the "flow over bridge" to be non-functional.

So I guess it's a bug, related to flow over bridge

supermerill, 2022-02-11:
use filament extrusion mult.
[TestFlow_BE.zip](https://github.com/supermerill/SuperSlicer/files/8051247/TestFlow_BE.zip)

@CCS86 it works well. Maybe you're confused because it create overlap instead of cahnging the E values.

CuredPrusa, 2022-02-11:
I wrote already: normal infill lines, perimeter lines, brim lines -those are fine, good bonding, not overextruded. Extrusion multiplier  works for all extruded plastic as far as I know. But I tested it anyway, And exactly as I thought, walls which were good before are missing width now.

[image]

CuredPrusa, 2022-02-12:
My personal conclusion after observation. 

Each time when Slicer has to automatically calculate...  
- Line pattern
- Flow intensity

...and that is always related to...

**1. Gap Fill
2. Bridge
3. Bottom
4. Top
5. Solid**

...the Slicer is calculating:
-	too much extrusion (flow)
-	too high line density

It could be that this is coming from Prusa.
It could be that this is not visible to users of "normal" small nozzles. Maybe magnitude of error is related to nozzle size. Maybe it's even beneficial with smaller nozzle, for better overlap. But with larger nozzle it's delivering unwanted overextrusion. Please note, I’m not even close to software developer, so I’m just guessing by observing the printer output. 

In my picture of "good" top layer is visible that there are simply too many lines in top fill. Impossible to handle with normal flow. Then I manually reduce the flow to 75%, in order to match too dense lines.

supermerill, 2022-02-12:
You're the only one having this problem.
If you can show me with numbers that the gcode is outputting too much plastic on solid areas vs perimeters, then I'll look into it again.

CuredPrusa, 2022-02-12:
Ok, I understand. 
I can't check the code to define this, I'm not that skilled. I think it's enough for anyone who is interested, to check my project, pictures, screenshots, description, it's easy to reproduce issue with attached project. 

However, I'm sincerely grateful to have ability to tweak flows (which are available), so I can find the sweetspot easily by testing separate segments of my object. The only two things that are missing for perfect control are
- Gap Fill flow control 
- Solid Fill flow control 

Thanks.

CCS86, 2022-02-12:
Okay, I see why you think it works, and I think it doesn't... that compensation is only being applied to infill lines, not to perimeters.

Here is a project that shows even with "above bridge flow" set to 200%, the layer above the bridge is identical to subsequent layers in line spacing, and in volumetric flow.

[Bridge Test.3mf.txt](https://github.com/supermerill/SuperSlicer/files/8054555/Bridge.Test.3mf.txt)

IMO, this "above bridge flow" compensation, would give a better result, if it were just a flow rate compensation applied to all lines over the bridged area. Why? Because the whole purpose is to recover a bridged area that has likely sagged. For the same reason that bridge flow is calculated very differently than normal flow (round extrusion vs thin rectangle), there isn't true support under the "above bridge" lines. With under constrained extrusions, printed with flow rates for a well constrained extrusion, you will just get spindly little lines, not likely to bond well to anything. Changing the line spacing to print more spindly lines doesn't seem like the best solution. 

A flow rate compensation, on the other hand, would help push extra filament into the under-extruded bridge, helping to fill these gaps and increase bonding with the bridge layer.

CuredPrusa, 2022-02-13:
I appreciate your effort.  
Yes, extrusion multiplier affects ALL extrusion which is happening during the print, regardless of line type.  

Right now I have to disable "gap fill" because at narrow sections of my object, too much gap fill (yes, this is what I'm getting - too much gap fill flow) protrudes through outer perimeter by means of deformation (not smooth surface as it would be without gap fill).  And I can’t do anything about it. My perimeters are fine, good measurements, except when middle section gets overfilled. I can’t have outer perimeter first because it makes nasty seam. 

Let's park my complaint about automatic calculation for a while. We're obviously getting different results and certainly having a different point of view. Let’s focus on my humble “new feature request” - selectable flow rate compensation for each line type. I can’t imagine how difficult would be to implement this in software. But that would solve all flow issues I’m experiencing. A simple test would show me the best settings for each line type. Then I select preferable flow for each project, save it in slicer’s project and get perfect and predictable print every time. It could be even related to filament type/manufacturer, experience will tell.

CCS86, 2022-02-13:
> I appreciate your effort. Yes, extrusion multiplier affects ALL extrusion which is happening during the print, regardless of line type.
>
> Right now I have to disable "gap fill" because at narrow [...]

This is the setting you should be trying first:
[image]

CuredPrusa, 2022-02-13:
@CCS86 
OMG, you're Gold !!!
Too much narrow focus blured my vision of bigger picture!
Thanks!

CuredPrusa, 2022-02-13:
Guys, please don't get me wrong. 
I'm just trying to contribute to make the best available slicer even better. That's all.

CCS86, 2022-02-15:
What speed are you guys printing gap fill?

Bowden or direct drive?

CuredPrusa, 2022-02-15:
[image]
In this example, Gap fill is moving from right to left (stopped before end of the line). 
I can say there is approx 15% more fill at the end of the gap fill line than at the beginning. After 11 conically expanding layers, the beginning of gap fill is almost flat (raised by maybe 0.05mm), while the end of gap fill goes by 0.15mm above general surface. It's hard to see on the picture, but this discrepancy was visible soon after layer 1, where gap fill was the thinnest line. 

Parameters:
Object width: 65mm
Filament PLA 1.75
Direct extruder E3D on Marlin
Nozzle 0.8
Layer: 0.32
Flow compensation: 1
Overlap Gap Fill: 0%
3 perimeters, 1.0mm
No changes in geometrical design of lines from bottom to top, but infill progressively increases width by each layer due to conical object. 
Perimeter speed: 25mm/s
Gap Fill speed: 20mm/s
Print order: from inner wall to outer wall, then gap fill. 
Temperature: 199C
Fan: 100%

Isn’t the purpose of gap fill to “fill the gap” and not to add more than the gap actually needs? Is there some reason why someone would like to overfill the gap and ruin the flatness of layer? Can I use negative value on Overlap Gap Fill parameter?

CuredPrusa, 2022-02-15:
Here's a piece of that gcode, if someone can see something from it...
[GapFillTest1.txt](https://github.com/supermerill/SuperSlicer/files/8074800/GapFillTest1.txt)
.

supermerill, 2022-02-16:
@tripp396

Maybe the tooltip isn't clear enough?

> This setting allows you to reduce the overlap between the perimeters and the gap fill.
> 100% means that no gaps are left, and 0% means that the gap fill won't touch the perimeters.
> May be useful to reduce if you can see the gapfill on the exterrnal surface

How can "no gap" vs " won't touch" can be done at equal flow?
maybe I should change it to

> 100% means that everything is filled, and 0% means that the gap fill is thin enough to  not touch the perimeters.

?

CuredPrusa, 2022-02-17:
I just want to brag here a little....
I have achieved a perfect top for my model!
This is not possible with quirky Cura, let alone Prusa with limited adjustments. 
Only SuperSlicer was able to do it right. 
I've made it in three layers:
1 bridge
1 solid
1 top

It's imposible to make or expect better with 0.8 nozzle!
Thank you Merill!

[image]
[image]
[image]