supermerill/SuperSlicer #2351: First layer height tooltip is at least badly worded

sprior, 2022-02-11:
### What happened?

When hovering over the First layer height field in Print Settings the last sentence reads "This can be expressed as an absolute value or as a percentage (for example 75%) over the lowest nozzle diameter used in by the object."

"used in by the object" probably means "used in the object", but unless we're talking about an IDEX situation with more than one nozzle can we really be talking about multiple nozzle diameters in a single print?  Are we really talking about the extrusion width?  It actually feels a bit strange to me (could be wrong or just used to S3D) that this percentage is of the nozzle diameter instead of a percentage of the base layer height.

But at the very least the wording of the tooltip is awkward enough to cause doubt in the mind of the user.

### Project file & How to reproduce

Hover over the first layer height field

### Version

2.3.57

### Operating system

Windows 10

### Printer model

Voron 2.4 with LGX

supermerill, 2022-02-11:
Can you propose a better one plz, so I can change it?

sprior, 2022-02-11:
I'd be happy to, but I'd need some questions answered to let me word it well:

- Is it really the physical nozzle diameter it's a percentage of or something else (extrusion diameter)?
- Can it really use multiple nozzles in that calculation?
- If there are multiple objects in the print do they all get the same value?
- If the printer has multiple nozzles but only one is used is only that one used in the calculation?
- Is it only the set of nozzles that's used for the first layer, or all the nozzles used at any layer in the print?

supermerill, 2022-02-11:
let me do a quick test.. (you could have done it yourself)
* yes, the nozzle diameter field from the extruders
* no, so it uses the smallest/thinnest
* depends on the extruder(s) used on each
* yes
* all layers, as I can't create the layer before choosing its height. I know, I may be able to exclude extruders that are much higher than all max layers...

qwewer0, 2022-04-07:
@supermerill
Could you explain in a little more detail why is it a percentage of the nozzle and not the base layer height?

I think the base layer height would be more logical to have this relation to the nozzle than the first layer height.

supermerill, 2022-04-07:
The first layer height if often fixed for an extruder diameter, as it need to be thick enough to not damage the print bed, and have good adhesion.

The base layer height is a more difficult to use as %, because of multiple extruders having different nozzle diameters and things using the base layer height as a stable baseline.

jrmann1999, 2023-02-08:
The issue with this, is when printing 0.3mm base layer height on a 0.6 nozzle, first layer is 0.45 at 75%.  This means the first layer is literally printing in the air above the bed.

Perhaps a sanity check to ensure the calculation doesn't result in a first layer that's larger in height than the base layer?  I've resorted to putting an absolute value in the field instead of calculation.