supermerill/SuperSlicer #2352: "Slice now" button gets covered by tooltip

Arakon, 2022-02-11:
### What happened?

When moving the mouse to the "Slice now" button, a tooltip pops up directly next to/above the mouse pointer.
[image]
 
It's often so close to the pointer, that even the few pixels of movement from clicking the mouse button can cause the user to click the tooltip instead, which does nothing, but also doesn't trigger the "Slice now" button.

Please adjust the position of the tooltip if possible, i.e. to appear completely off-button.

### Project file & How to reproduce

[blank.zip](https://github.com/supermerill/SuperSlicer/files/8046738/blank.zip)

### Version

2.3.57.10

### Operating system

Windows 10

### Printer model

-

supermerill, 2022-02-11:
The tooltip is made by the os, I can't do much about it.
I can make this one a one-lin, so it's more often below the mouse cursor vs above?

Arakon, 2022-02-11:
Oneliner might work. I'd prefer not to disable tooltips globally, as they can be useful.

closed 2022-04-01 by supermerill

markus-i, 2022-10-02:
Not sure whether I should comment on this issue (as you already proposed a solution in your first post above) or open a new issue - if you prefer me to open a new issue, please tell me.
My problem: the tooltip for (some of) the modifier settings is more or less completely covering the setting I want to change:
[image]

So, yes, please - add a preference to hide tooltips completely. Put it behind a few bariers, if you want (for example, hide it in Expert mode). It's not a bug, but an annoyance - which gets more and more annoying the more modifiers I am using. I'm now at ~20 modifiers and counting...