supermerill/SuperSlicer #2353: Inwards slope sagging issue with low infil percentage.

levleon13, 2022-02-11:
I don't know how to define this post, not sure what would be the solution as there's already one implemented that doesn't work with
petg which is very liquidy. Solid infil ( pink)  does not do the job to prevent from the sloping perimiters to sag inside the model.
Adding 4th or 5th perimiter will definitely help, but is huge overkill, overweight, and over print time, to just tackle this small area ( 2cm squared area in a 30cmX20cm print ). What would you suggest to do in this case? 

[image]
[image]
[image]
[image]

supermerill, 2022-02-11:
Please send a correct project file created via "save prject as"

levleon13, 2022-02-11:
> Please send a correct project file created via "File -> save project as"

[mid.zip](https://github.com/supermerill/SuperSlicer/files/8048197/mid.zip)

supermerill, 2022-02-11:
set `no solid over ... perimeters` to 0 to deactivate it.

levleon13, 2022-02-11:
> set `no solid over ... perimeters` to 0 to deactivate it.
i did, and i still have the pink solid infill under the slope. But pardon me, isn't that mean i am now will have even less "ground" to lay the sloping perimiters onto, and they will sag even more?

supermerill, 2022-02-11:
That means you will have more solid infill (purple) near the slope to reinforce them.

You can also activate the 'infill_dense' to better support falt area, I can see one or two places where it can be useful.