supermerill/SuperSlicer #2354: Relocating Retraction Settings

FawkesOSU, 2022-02-11:
**Is your feature request related to a problem? Please describe.**
The current location of retraction settings is not an "issue" but more of a limitation. Where they currently are, it forces you to either run a much more aggressive retraction than needed for certain materials due to the settings needed for others, or you must change these settings with every change of filament. 

**Describe the solution you'd like**
If the retraction settings were moved to the filament settings and were saved to each filament rather than the extruder, this limitation would be fixed. 

**Describe how it would work**
I imagine the filament settings could be implemented in a similar fashion to the temperature settings. 

**Describe alternatives you've considered**
As previously mentioned, running overly aggressive retraction, or manually swapping settings each time is the current solution. 

**Additional context**
Any settings that change with filament should fall under the filament selection. Temp, retract, lin-adv (or similar)  all of these should be filament settings.

supermerill, 2022-02-11:
look at the "override" section fo the filament settings

sebastian-be, 2022-03-23:
@FawkesOSU, if satisfied with solution, please close ticket.

closed 2022-03-23 by FawkesOSU

Shine753, 2024-08-28:
In v2.5.59 of SuperSlicer, there are 2 places for retraction settings : one for the extruder, one for the filament.
How is this supposed to work ?