supermerill/SuperSlicer #2356: External_Perimeter_Cut_Corners  - simple parameter improvement

CuredPrusa, 2022-02-12:
### Version
Version 2.3.57
Build: SuperSlicer_2.3.57.10

### Behavior
I see the benefit of "Cutting Corners" function, I print some strong and precise mechanical components. . But with just 5% flow reduction at 100% setpoint, it's barely visible with 0.8 nozzle. 
It would be beneficial if you can scale 100% value to 10% flow reduction. Even if it would be too much, there is a fine tuning in percentage available, where 50% would be the same as it is now at 100%.

Thanks

supermerill, 2022-02-12:
100% remove the exact theoretical extrusion amount.
Changing that, remove any meaning to the percentage, no?

CuredPrusa, 2022-02-12:
I understood that value 100% is removing 5% of full flow at the corner. 
All I'm saying is it would be nice to have more. 10% removed at value 100%. 
If it would be too much, I could trickle down in percentage.
  
old scaling: 0=0% removed - 100=5% removed
new scaling: 0=0% removed - 100=10% removed

supermerill, 2022-02-12:
The value depends on the angle (Note that I added a cutoff for when it's too shallow). The 5% you quote is only applicable for a defined angle.

CuredPrusa, 2022-02-13:
Ok, now it's clear.... that I'm oversimplifying!
My observation is that corners could be more trimmed at 0.8 nozzle. 
I hope you do count on filament flow inertia which is quite high with big size nozzle.

supermerill, 2022-02-13:
Iif the extrusion is bigger, there is a bigger area to remove, but the flow is also higher, so it's the same %.

> I hope you do count on filament flow inertia which is quite high with big size nozzle.

I don't do pa estimation nor speed estimation.
I remove the flow at the start of the next extrusion, just at the turn.

CuredPrusa, 2022-02-14:
I'm not even trying to comprehend your level of knowledge. I'm just saying it would be nice to trim the corners even more, if possible. I'm just a grateful user, that's all. 
Thanks.