supermerill/SuperSlicer #2358: New feature request - adjustable infill area threshold

CuredPrusa, 2022-02-12:
### Version
Version 2.3.57
Build: SuperSlicer_2.3.57.10

_Is this a new feature request?_
Yes!
It would be very usefull to control where should infill be placed, by means of selecting minimum infill area , below which the space is left hollow or filled solid (toggle options)

supermerill, 2022-02-12:
So you ask for a sitwh for the current `Solid infill threshold area` so itwon't fill instead of solid?

CuredPrusa, 2022-02-12:
Yes, something like that.

Screenshot is showing one half of upwards expanding conical tube. 

[image]

There is some minimum area threshold for infill/solid already in the software. At layer 58 (and lower), neither infill nor solid gets generated -by default, hardcoded in software (the area is too small)

[image]

In Prusa, filling goes all the way in small slot, leaving just tiny dots of fill to print in the end, and that is ridiculous! However, I’d like to change that hardcoded threshold, both for solid and infill. Sometimes I need solid in narrower gaps ("gap fill" is not useful here, because it gets everywhere). Sometimes I need more of hollow space and I’d like to “tell” infill to start at larger hollow surface.  Sometimes smaller hollow surface needs infill. 

[image]

PERFECT slicer would have ability to DRAW the areas for infill and solid, similar to drawing the support. But let’s skip that for now.

supermerill, 2022-02-12:
> PERFECT slicer would have ability to DRAW the areas for infill and solid, similar to drawing the support. But let’s skip that for now.

you can add modifier geometry and set your % infill to 100% inside.

CuredPrusa, 2022-02-13:
Thank for a tip. But I can't find info about "modifier geometry" to learn about it. What is it?

supermerill, 2022-02-13:
basically https://help.prusa3d.com/en/article/modifiers_1767

CuredPrusa, 2022-02-14:
Wow!

supermerill, 2022-02-14:
also for your model, you can set your threshold to 0% and your infill ratio to 0% if you don't want them.

CuredPrusa, 2022-02-14:
Considering the example presented by those two screenshots, I would actually like to be able to adjust parameter which is enabling infill to layer 58 and 57. Those are now too narrow sections for default infill generator .  (Cone is expanding upwards)
That's what this topic is all about. 

I'll dive into "modifiers" asap, that looks very useful and powerful!