supermerill/SuperSlicer #2360: Internal bridge infill direction bug

wtfiAndrewNelson, 2022-02-12:
**Bug description**
Internal bridge infill direction is the same as the support (when using combined) below it making it sag

**Expected behavior**
Internal bridge infill direction should be 90 degrees to the (rectilinear, combined) support layer beneath it

!!>> Project File <<!!
[Impeller.zip](https://github.com/supermerill/SuperSlicer/files/8054148/Impeller.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Slice with infill combine every 2 layers (anything >1)

**Screenshots**
As seen from below:
[image]

**Desktop (please complete the following information):**
I have replicated it on all of my windows machines

**Additional context**
If combine infill setting is used at a value greater than 1, with rectilinear infill, the internal bridge support will run the same direction as the rectilinear support below it, making it sag.

supermerill, 2022-02-12:
You enforce your bridge direction 90° from the fill angle, so of course it's aligned (the direction change by 90° each layer).

wtfiAndrewNelson, 2022-02-12:
Sorry, I sent you one of the hundreds of different tries to get this to work and I don't know which one. However, they all have worked correctly when I reduce the "combine infill every" to 1 layers. I am sorry for the confusion but if your try to recreate it instead of looking for what I did incorrectly, you should find the same situation. I have done this with automatic angles and it does the same.
[image]
[Impeller.zip](https://github.com/supermerill/SuperSlicer/files/8054511/Impeller.zip)

With the above, it is making the lines parallel instead of at a 90 degree angle to one another, however, if I change the combine layers back to 1, everything is fine.

wtfiAndrewNelson, 2022-02-12:
[image]
[image]

wtfiAndrewNelson, 2022-02-12:
These are from below and the only change is combine every (1 or 2) layers

wtfiAndrewNelson, 2022-02-13:
Okay, thanks. I have a workaround and just wanted you to know about this behavior.

supermerill, 2022-02-13:
hey, the behaviour of the fill angle isn't right. It's still a bug, no need to close it until it's resolved.