supermerill/SuperSlicer #2361: Force retract on first layer

tome9111991, 2022-02-12:
Hi
if the option in infill -> Only retract when crossing perimeters enable.
i have some missing retract on first layer durring travel over solid infill.

enable etract when crossing perimeters:
[image]

disable etract when crossing perimeters:
[image]

Maybe there is a way to force retract on first layer only with enable this option?

supermerill, 2022-02-13:
> if the option in infill -> Only retract when crossing perimeters enable.
i have some missing retract on first layer durring travel over solid infill.

what are the problematic travels? Can't really spot them.

tome9111991, 2022-02-13:
for exemple on this travel:
[image]

halfway I lose plastic and it sticks to the bed. (because no retract before this travel)
the solid infill then moves over the plastic on the bed and gets artifacts on the first layer

supermerill, 2022-02-13:
Ok. A little 'not on first layer' choice next to " Only retract when crossing perimeters" should do the trick?

tome9111991, 2022-02-13:
> Ok. A little 'not on first layer' choice next to " Only retract when crossing perimeters" should do the trick?
> 
Yes sounds good

> Only the first or the X first layers?
I think only the first layer is good.
But know maybe the value from bottom layer are good too

tome9111991, 2022-03-20:
I was thinking about this issue.

A  'not on first layer' for the X first layers is better choise .

supermerill, 2022-03-25:
> Maybe a Only retract when crossing perimeters" not for Solid infill is the best way.

so only retract when the travel only goes over some internal perimeters and sparse infill?
and stop if crossing an external perimeter, or a solid infill?

then i guess i can remove the "enforce on first layer", as it's mostly solid infill.

tome9111991, 2022-03-25:
> > Maybe a Only retract when crossing perimeters" not for Solid infill is the best way.
> 
> so only retract when the travel only goes over some internal perimeters and sparse infill? and stop if crossing an external perimeter, or a solid infill?
> 
Yes, i need retract on external perimeter and solid infill.

tome9111991, 2022-03-26:
> > I need an extra option to have retract on first layer and solid infill
> 
> but there is almost only solid infill on the first layer, why adding this constraint?

you are right.
only need for solid infill

emailcraigs, 2022-05-07:
Will a force retract on first layer be implemented? I'm having same issue as OP.

closed 2022-07-21 by tome9111991