supermerill/SuperSlicer #2369: Transparent modifiers and supports

robustini, 2022-02-13:
It would be very useful if the modifiers when repositioning or resizing were as transparent as unselected ones.
That way it would be much easier to position and resize them correctly based on how the STL is made.
Same with supports.

[image]

supermerill, 2022-02-13:
I don't really understand.
They should have the same "Frame" (origin, orientation, scale) as the part, so If you created with your 3D modeler, it should be at the right place.

neophyl, 2022-02-13:
I made a very similar feature request on the PS github ages ago.  https://github.com/prusa3d/PrusaSlicer/issues/3678 or https://github.com/prusa3d/PrusaSlicer/issues/4438 , as when placing modifiers added via the software they aren't done in CAD.  People use the built in shapes to get extra perimeters around holes for example and to avoid surface discontinuities it can help to sink the modifier below the outer surface so that it looks better on the outer layers.  This can be problematic as once its inside then manipulating it or judging its position is much more difficult.

Also a secondary side benefit of a transparency would be to show if you have one part inside another on the plater.  When adding several parts all at the same time if you have very small ones its possible to miss them as you cant see them.  I've got into the habit of selecting parts one by one on the right hand panel and looking for them in the window to make sure I don't do that but I'm not the only one who has that problem.  There have been a few threads on the forum where its cropped up too.

I do understand that changes to the gui like this can be difficult which is why I was hoping that PS could throw resource at it :)

robustini, 2022-02-14:
> I don't really understand. They should have the same "Frame" (origin, orientation, scale) as the part, so If you created with your 3D modeler, it should be at the right place.

I'm, sorry, maybe I expressed myself wrong, I meant a sort of x-ray when positioning and resizing, like the two feature requests of the PS GitHub.

closed 2022-04-04 by robustini