supermerill/SuperSlicer #2373: No "Thin Walls" and No "Overlapping External Perimeter" + Thin Model = Crash

jhlchu, 2022-02-14:
### What happened?

## What is the problem?
Slicer crashes when "Thin Walls" and "Overlapping External Perimeter" are uncheck and slicing a thin model.

## Steps to reproduce
1. Uncheck "Perimeter & Shell" > "Overlapped External Perimeter"
2. Uncheck "Perimeter & Shell" > "Thin Walls"
3. Set "Width & Flow" > "First Layer" > "Width" : 0.45
4. Go to "3D View" > "Add Shape" > "Cube"
5. Set Cube dimensions as 0.40 x 25 x 25 (XYZ)
6. Slice

## Expected outcome
An error of some kind error to allow user to recheck option.

### Project file & How to reproduce

n/a

### Version

2.3.57.10

### Operating system

Win 10 Enterprise

supermerill, 2022-02-14:
maybe the brim bug? can you try the 57.11?

jhlchu, 2022-02-15:
Tried with 57.11, still crashed.
Also crashes on SS 2.3.6.9 and SS 2.3.57.9.
Opened the .3mf in PrusaSlicer 2.3.3 and 2.4.0, and sliced. Both give the error: "The print is empty. The model is not printable with current print settings" instead of crashing out like in SS.

[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/8066230/Shape-Box.zip)

closed 2022-07-14 by supermerill