supermerill/SuperSlicer #2377: Support enforcers not enforcing

Arakon, 2022-02-14:
### What happened?

I am quite certain this USED to work, 2 or more versions back.
When using "Supports on build plate only" and painting on support enforcers, it would originally autogenerate build plate supports, plus the ones painted on. However, it no longer does and ignores the paint-on supports completely.
[image]
[image]

When using "Support enforcers only", those are successfully generated.

### Project file & How to reproduce

[MinisatsanaM3x4 inserts.zip](https://github.com/supermerill/SuperSlicer/files/8061866/MinisatsanaM3x4.inserts.zip)

### Version

2.3.57.11 Final

### Operating system

Windows 10

### Printer model

Ender 3

neophyl, 2022-02-14:
Took a look at the project, you have supports on build plate only selected. As such it can’t support that top area as it would have to put supports on he part and your option doesn’t allow that. 
When you change for support enforcers only then limit is no longer applicable so it’s now allowed to build support on the part so it does so. 
Personally I can’t see anything wrong with the behaviour.

Arakon, 2022-02-15:
I had figured that support enforcing overrules that for the painted areas.
Problem is then, I'd only have the option to have supports everywhere, which would create supports in that area anyways, and I'd have to block supports on the entire inside of the model, or have supports on build plate only, in which case I can't get supports on that small area. 
If that is intended behavior, then maybe consider this a feature request, to have support on build plate + enforcers as an option.

neophyl, 2022-02-15:
Its my understanding that if you choose support everywhere that the support locations are chosen automatically and determined by the overhang angle you have specified.  So it merrily goes about and creates support all over the place.  The software is making the decisions.

The support on build plate only option just limits it to start from the plate and not on the model but the defined overhang angle still determines areas that it thinks it needs to support and its still automatic.

For Enforcers only, you, the intelligent human being are choosing the areas that you want supporting.  So we have the human in the decision making process.  As such its up to you to define where needs support.  If your decisions mean it has to put support on the part then that is up to you, its just trying to fulfil your requests for the areas you have marked.  As its very easy to define areas to enforce then this is the method I most commonly use now. 
Cant remember the last time I used a blocker or support everywhere.  Its just too indiscriminate.

Besides just wait till the PS 2.4 is merged in and the 2.4 supports are available.  Game changer when it comes to how usable the supports are.

neophyl, 2022-02-18:
Imply is such a loose word though :)  That whole sentence is worded very muddily.  There whole support process is 2 basic steps.  The first is marking areas that need support.  With Auto you define those areas with the overhang threshold.  That paragraph means that you can define further areas using the paint on methods to add additional ones that setting a small overhang threshold doesn't cover.  Of course you can also block out large areas with a enforcer modifier too.  Its all just defining areas on the model that do or do not need support.

Once that process is done though the support algorithm takes over.  It will generate supports, BUT if will still obey the settings you have configured.  Bed only,all over, spacing, angle etc etc.

Try it with a PS version 2.3.x and they generate support exactly the same way as the Susi versions you are using.   If you paint an area above part of the model and have From build plate only enabled it does not produce the support.  That's the support for 2.3 based versions.

Now Prusa changed the supports behavior in 2.4, for 2.4 it DOES create the supports when you paint them on even if you have from build plate only.  The painted areas do in effect over ride the from build plate only.

The Prusa web pages are also not very specific, as they are updated for new released versions and they assume that you are using the latest version.  If you are using an older version then they don't always make sense.  They don't version them so that you can look up instructions for say 2.3, or ones for 2.4.

neophyl, 2022-02-18:
Supermerill is currently working on building a version that merges in the PS2.4.1 code.  Given how much was actually changed in PS 2.4 its not an easy job.  I suspect a lot of the changes will be breaking his stuff.  

There's also what to do with duplicate functionality, SuSi has had lots of features that weren't in PS and now PS has added a form of some of them too. 

There is a beta 2.4 on the releases page https://github.com/supermerill/SuperSlicer/releases but as mentioned there its pretty raw still and you can expect issues.

hany, 2022-02-18:
Gotcha, thanks for the info. Just learned Susi = SuperSlicer :)