supermerill/SuperSlicer #2385: Gcode preview will not render if switched to too early.

levleon13, 2022-02-14:
Slicing an object once - shows GCODE preview 
Any further changes on the object - no GCODE preview anymore ( empty bed preview ).
it does show up if i move the object multuple times ( about 5 times of rerender to get gcode preview again ).
My program is set to generate gcode automatically which worked perfect till this moment.

Expected to be rendered each change ( as it did before my fresh app install ).
Also found a behaviour pattern. It will 100% show gcode if i will wait for it to complete the silcing on the 3D view tab. if i will switch to gcode preview WHILE it is slicing, it will not show it.

[front bridge.zip](https://github.com/supermerill/SuperSlicer/files/8063250/front.bridge.zip)

[image]

supermerill, 2022-02-14:
> Also found a behaviour pattern. It will 100% show gcode if i will wait for it to complete the silcing on the 3D view tab. if i will switch to gcode preview WHILE it is slicing, it will not show it.

understood, yes, it should refresh.
if shows up if you click on sliced preview then gcode preview, right?

> Any further changes on the object - no GCODE preview anymore ( empty bed preview ).

Yes, I finally fixed the bug that kept old data. It should move you to the 3D view if you have activated the auto-switch to preview.

levleon13, 2022-02-15:
Once again, you are the best. Thanks!