supermerill/SuperSlicer #2386: Crash When Displaying Volumetric Flow rate

CCS86, 2022-02-14:
### What happened?

SuSi crashes when I choose to view the volumetric flow rate in this project.

### Project file & How to reproduce

[OP10-Fix2 .3mf.txt](https://github.com/supermerill/SuperSlicer/files/8064196/OP10-Fix2.3mf.txt)

### Version

2.3.57.11

### Operating system

Windows 10 x64

supermerill, 2022-02-14:
It's only with this project?
Can't reproduce.
[image]

CCS86, 2022-02-14:
Hmmm, weird. It crashed like 4 times in a row. Then I recreated again, saved the project, switched to volumetric flow and it crashed again. Thought for sure I caught it. 

Not sure what changed on my end, but now it works for me too...

supermerill, 2022-02-14:
maybe not enough memory?