supermerill/SuperSlicer #2388: Wrong slicer name and version in first line of G-Code

MrDix, 2022-02-14:
### What happened?

The generated G-code starts with 
`; generated by PrusaSlicer 2.4`
instead of 
`; generated by SuperSlicer 2.3.58.0`

The result is that Mailsail and Fluid show the wrong slicer in the gcode list:
[image]

### Project file & How to reproduce

Here is a generated g- code from SuperSlicer 2.3.58.0 for example:
[2.4.58.0_midspan_panel_clip_3mm_x12-4.01974g-16m.gcode.zip](https://github.com/supermerill/SuperSlicer/files/8064552/2.4.58.0_midspan_panel_clip_3mm_x12-4.01974g-16m.gcode.zip)

### Version

2.3.58.0

### Operating system

windows 10

### Printer model

Voron 2.4 with Galileo Extruder

supermerill, 2022-02-14:
haha.
Prusa created it 3 years ago by using the generic 'SLIC3R_APP_NAME'
And now they decided to hardcode it to PrusaSlicer.

mmm... I guess if some other tools use it, it's maybe better to fix it to a defined key...
I wonder if it's just a display of what's after the "generated by" or it needs a specific name.

MrDix, 2022-02-15:
This string is interpreted/checked somehow. It only seams to accepts `Slic3r`,`PrusaSlicer` or `SuperSlicer` and it requires a version number (int or float) directly afterwards. If you write something before the known slicer string it is ignored.

closed 2022-07-14 by supermerill