supermerill/SuperSlicer #2398: Avoid crossing perimeter (is still crossing perimeter)

TommyC81, 2022-02-16:
### What happened?

I'm using setting "Avoid crossing perimeter" with "Max detour length" set to 0 (zero) - this should disable the detour length consideration and simply avoid crossing the perimeter all the time. However, I'm getting lots of perimeter crossing in places where I'd expect it to be avoided.

Example project and images attached.

### Project file & How to reproduce

[Avoid perimeter crossing - example.zip](https://github.com/supermerill/SuperSlicer/files/8077176/Avoid.perimeter.crossing.-.example.zip)
[image]
[image]

### Version

2.3.57.10

### Operating system

Windows 10

### Printer model

Creality CR6

Pavulon87, 2022-02-16:
It looks like this has stopped working since SuperSlicer_2.3.57.7
2.3.56.6:
[image]
2.3.57.7 and newer:
[image]

TommyC81, 2022-02-20:
[image]
[Avoid perimeter crossing - example (nightly 2022-02-19).zip](https://github.com/supermerill/SuperSlicer/files/8103648/Avoid.perimeter.crossing.-.example.nightly.2022-02-19.zip)

To confirm, the avoid crossing perimeter works better in the nightly from 2022-02-19. However, note the repeated perimeter crossing that I highlighted in previous post is still occurring. Image and project file attached.

supermerill, 2022-03-31:
The new algo from 1.3.57.7 need a bit more width (#1794), hence why it's now failing at this object ( it needs to be scaled by 140% to recover good behavior).

The algo should fallback to 'old' behavior of following the external periemter whan it's too thin, I'll have to investigate a lot to find why it's failing.

Ygle38, 2022-05-10:
Hello, 
Just another case of crossing perimeter, with supports.
- this part is ok with no crossing (in case of no support), but adding support reveal the failure; travelling from left to right support by crossing the main part.
- probably the same issue, I post it in case it helps to understand the code regression
[image]
[Vtol1.zip](https://github.com/supermerill/SuperSlicer/files/8663847/Vtol1.zip)

c2v4, 2022-05-17:
Yeah, I have the same issue. I found out that this only shows up for a certain range of travel:
![image](https://user-images.githubusercontent.com/11049808/168752393-1afa110c-7aff-425f-8d34-711fa
[k40exhaust-Body1-1.zip](https://github.com/supermerill/SuperSlicer/files/8706093/k40exhaust-Body1-1.zip)
69b0fb0.png)
In layers that are close to those that are affected by it, infill join is in the same place as well as the 'turn point' of internal perimeter. I've attached my project file in case that would help you find the answer. SS Version: 2.4.58.2