supermerill/SuperSlicer #2401: Eject SD

Ablast2bme, 2022-02-16:
### Version
2.3.57.11
### Operating system type + version
Windows 10 Version 21H2

### Behavior
* _Describe the problem_ After inserting a SD card and exporting a file to that SD card a dialog box appears on screen. The box reads   Exporting finished, Open Folder and the name of the file exported to the SD card below that.  It also has two buttons.  One is to eject SD card and the other is to close the dialog box. The close button works correctly but the eject button does not. It States that ejecting of the devise has failed. And it has a close button that works correctly.
* _Steps needed to reproduce the problem_ Insert Sd Card and Export a file to it and then try to eject it using eject button.
* _Expected Results_ It should Eject make the SD card removable with out  having to anything else. 
* _Actual Results_ States that ejecting of the devise has failed
  * _Screenshots if useful_ Not sure if needed

_Is this a new feature request?_ No
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

supermerill, 2022-02-16:
95% change it's sue to your os / usb key.

Can you reproduce with prusaslicer 2.3.3 ?

Ablast2bme, 2022-02-17:
It does not work using Prusa slicer but works with USB card reader. It's not a big problem, just thought I would report it. 
Also tried a different Dell Laptop with a Full-size SD Slot and it does not work.

sarvenn, 2022-02-17:
@Ablast2bme is it similar to this? I had reported an ejection issue on PS2.3 and 2.4 as well, however it seemed to be an unique issue for a very restricted user due to their card reader hardware. On the other hand I could eject sd card without any problem from by Cura.
https://github.com/prusa3d/PrusaSlicer/issues/5734