supermerill/SuperSlicer #2403: Only retract when crossing perimeters   -a little bit more tuning...

CuredPrusa, 2022-02-17:
### Version
Version: 2.3.57.11
Build: SuperSlicer_2.3.57.11

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

### Behavior
The perimeter crossing should be a bit more defined in coding engine. The nozzle is not quite away from perimeter, so it doesn't retract. But with big nozzle (0.8) it is just enough to leave artefacts on the surface of print. What red arrow is showing, is actually visible on the surface. Model is conical, that's why it is changing travelling positions progressively. I don't think some manual adjustment availability is necessary here. Maybe a change of threshold in hardcoded software would be just enough.  

[image]

[image]

And while we're at it, I have one pragmatic concern. 
In the upper left corner is shown that movement line is blue, retraction line is red color (if I didn't misunderstand). But this color change does not happen on preview, so I have to look at retraction dots to actually see what is travel with retraction and what is travel without retraction. Can this be fixed somehow?

Thanks!

supermerill, 2022-03-31:
can you post the project file?

CuredPrusa, 2022-03-31:
Can i sent it to you through PM in discord?

supermerill, 2022-03-31:
yes

CuredPrusa, 2022-03-31:
OK, here it goes. 
Affected layers are 
46.72-49.92
83.84-87.04

CuredPrusa, 2022-03-31:
I was trying to cut it at the first place, 
but I gave up 
because for some strange reason slicer didn't reproduce the problem (the same travel path) on cut out version.