supermerill/SuperSlicer #2406: Nigthly windows x64 #539 starts and stops immediately

robustini, 2022-02-17:
I also tried to delete the SuperSlicer folder in Roaming, it regenerates it at the start but then exits immediately after the start screen.

robustini, 2022-02-17:
Same issue with the accel edition "2.4.58.0".

RazorCopter, 2022-02-17:
same problem SuperSlicer_2.3.58.0_win64_220214

MLiv79, 2022-02-18:
Same issue here, after the first run, superslicer won't start anymore

tome9111991, 2022-02-18:
fix windows crash at startup nightly is starting.
bug got a white space:
[image]

robustini, 2022-02-18:
> fix windows crash at startup nightly is starting. but got a white space: [image]

Yes, there's white space.

supermerill, 2022-02-18:
Found the problem. FIxed in nightly. Will update the labels soon.

robustini, 2022-02-19:
> 

Thanks!
Please also check the add enforced support, if I insert them a block support appears instead.
And the drop down menu inside add block support is not selectable.

robustini, 2022-02-20:
@supermerill is there a chance to remove this annoying starting waring?

[image]

supermerill, 2022-02-21:
for the annoying starting warning, I'll see where I can add a var to disable it after first show.

supermerill, 2022-02-21:
The modifier will be fixed (note that you can change their type afterwards)

robustini, 2022-02-21:
> The modifier will be fixed (note that you can change their type afterwards)

If you go to "change type" it already appears as an enforcer, but in the list it is put as "Blocker-Box", not "Support-box".

[image]

supermerill, 2022-02-21:
yes, the position of the seam-enforcer was switched.

robustini, 2022-02-27:
@supermerill the trick of the "NO_DLL_WARNING" dummy file only works if you open SuperSlicer from its executable, but if you open an STL from Windows with the "Open with" option it doesn't work, the message always appears.
So maybe that file should be placed in the folder inside Roaming, and the check done on that directory.

supermerill, 2022-03-01:
ok

robustini, 2022-03-04:
@supermerill I also report a problem with the fan, it does not respect what is set.
100% in the top layers and instead turned it off, 40% desired speed and put it at 86.

supermerill, 2022-03-07:
> I also report a problem with the fan, it does not respect what is set.
100% in the top layers and instead turned it off, 40% desired speed and put it at 86.

please open a new issue with the project file for that.

robustini, 2022-04-04:
The problem exists and it's boring, I have the ssd full of NO_DLL_WARNING, everywhere! ;-)

supermerill, 2022-04-04:
the code has been moved. Now it's hidden via a preference entry.

robustini, 2022-04-04:
@supermerill if the fixed version is the one compiled three hours ago now the situation has worsened, the warning always appears... :-(

closed 2022-04-04 by robustini