supermerill/SuperSlicer #2412: Default fan speed at 100% results in M106 S100 - not M106 S255

McNugget6750, 2022-02-18:
### What happened?

When setting the default fan speed to 100%, the slicer produces an M106 S100 gcode from it, instead of M106 S255.

### Project file & How to reproduce

[15mm hole calibration_v2_Fan Speed PWM Issue.zip](https://github.com/supermerill/SuperSlicer/files/8093929/15mm.hole.calibration_v2_Fan.Speed.PWM.Issue.zip)

### Version

2.4.58.0

### Operating system

Windows 10

### Printer model

Customized Malyan M180 running Klipper and Fluidd

closed 2022-02-18 by McNugget6750

supermerill, 2022-02-18:
> Not sure why this is set by default, though...

You can see the default value in the tooltip, it's default to false.
Maybe you are using a profile of someone else that needs it or the base printer needs it.

McNugget6750, 2022-02-18:
Hm, that's odd. I did a fresh install and created a custom printer using the config wizard. I did not use another person's config or project - I also did not check this box. I have no explanation how this was set on my end.