supermerill/SuperSlicer #2416: Duplicating changes in multi extruder config

Pavulon87, 2022-02-19:
**Is your feature request related to a problem? Please describe.**
When working with multiple extruders, it would be good to have the option to change the settings of all extruders at once.
At the moment, if we want to change one option (enable or disable retraction, wipe, z-hop, etc.), we have to manually repeat it for each extruder.
It would be good to have a switch that would enable/disable the clone settings for all extruders.

**Describe how it would work**
This could work just like for "nozzle diameter", except there would be an option to enable/disable.
[image]

supermerill, 2022-02-19:
If you're using the same filament on each, you can use the filament override.
If there are all identical, you can remove them, modify the first and re-add them.
I have a hard time to see when you have to change something on all of them unless it's the first setup, at preset creation.

Pavulon87, 2022-02-19:
Filament override looks like what I needs. Sometimes I test different multicolor print settings and it takes time to change all the colors.

closed 2022-02-19 by Pavulon87