supermerill/SuperSlicer #2417: Option to ignore "compatible printers condition"

Pavulon87, 2022-02-19:
**Is your feature request related to a problem? Please describe.**
Since using SuSi I have created a lot of .3mf files saved with the default "compatible printers condition" settings.
But now I have other printers too, I have to remember to delete this value every time because the settings are not compatible.
[image]

**Describe the solution you'd like**
Maybe adding an option in Preferences to ignore compatibility wouldn't be too problematic.

**Describe how it would work**
There is already an option "Show incompatible print and filament presets" but this option would also add an option to ignore compatibility when changing printer.
[image]

Pavulon87, 2022-03-17:
I added `return true;` at the beginning of all functions `is_compatible_with_print` / `is_compatible_with_printer` in the file **Preset.cpp** and it seems to be working well.
Is it possible to use `if ( wxGetApp().app_config->get("show_incompatible_presets")[0] == '1' ) return true;` in this file without big modifications?

supermerill, 2022-03-18:
not possible, as Preset is inside the slic3rlib, and the show_incompatible_presets is from the gui, and the gui use the lib and not the opposite.