supermerill/SuperSlicer #2418: Lost screen when flagged "need for slicing"

CuredPrusa, 2022-02-19:
### Version
Version: 2.3.57.11
Build: SuperSlicer_2.3.57.11
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

### Behavior
When I adjust the additional gcode generator (layer column on right), I work in Sliced Preview because in Gcode Preview I lose preview everytime I change something, which makes workflow very slow. Ok, at least I have Sliced Preview. But when I change something in Height Range Modifier, I lose object each time, even in Sliced Preview. Can it be done the same as when I add gcode, to wait until I press "Slice now"? 

In fact, I think it would be wiser to treat all changes like this: keep image of object available all the time (with old parameters). Of course I have to Slice every time I change something, to see actual changes on screen. But please let me finish all the changes in peace.     

_Is this a new feature request?_
YES

supermerill, 2022-02-19:
what are your settings in the automation seciton?
(anto-center, background processing, switch to preview)
Because I don't see the problem with mine.

CuredPrusa, 2022-02-19:
Switch to Preview is set to Don't switch.
Any other setting there is auto switching to **3D view**, which is not useful because I lose overview of layers and scrolling through layers.
Background process is OFF because I don't want to be disturbed when changing many inputs at once. I rather press Ctrl+R when I'm ready. I understand if **Gcode preview** can't work without slicing, but at least I'd like to stay steady on **Sliced preview**, similar as when I add (or modify) custom gcode on the scroll bar.

Each time i change some value in Modifier, I lose overview of layers. Then I have to slice again.  I have to see layers, to modify many layer heights at many instances (Height range Modifier). 

Now it is like this:
one value change = one slicing = annoying.

supermerill, 2022-02-19:
I guess i can add a "Don't refresh until next slicing"