supermerill/SuperSlicer #2419: EXTERNAL PERIMETER FAN SPEED adjustability for other perimeters.

CuredPrusa, 2022-02-19:
### Version
Version: 2.3.57.11
Build: SuperSlicer_2.3.57.11
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

### Behavior
Please make an option for EXTERNAL PERIMETER FAN SPEED to relate to all perimeters, not just one on the outside. 
I want infill to go with less cooling, but then inner perimeters are overheating. When nozzle reaches outer perimeter, it's already too late. 

_Is this a new feature request?_
YES

supermerill, 2022-02-19:
Can't add all possible fan seed override with the actual code. Have to rewrite it so I have a good base.

jhlchu, 2022-02-20:
Behavior possible using "Between Extrusion Role G-Code Change"

```
; External perimeter
{if extrusion_role=~/ExternalPerimeter/};[extrusion_role]
M106 S255
; Perimeter
{elsif extrusion_role=~/Perimeter/};[extrusion_role]
M106 S255
; Internal infill
{elsif extrusion_role=~/InternalInfill/};[extrusion_role]
M106 S0
; Top solid infill
{elsif extrusion_role=~/TopSolidInfill/};[extrusion_role]
M106 S0
; Solid infill
{elsif extrusion_role=~/SolidInfill/};[extrusion_role]
M106 S0
{endif}
```

and -1 for fan speed in filament settings

CuredPrusa, 2022-02-22:
If this will be implemented someday, please add it to Modifiers as well.