supermerill/SuperSlicer #2420: Internal Bridge Infill instead of Solid Infill

Sirnut, 2022-02-19:
### What happened?

When slicing an STL, "Internal Bridge Infill" is generated when I believe regular Solid Infill should be

In the attached files, you can see layers 2-6 are Internal Bridge Infill instead of Solid and Top Solid infill

### Project file & How to reproduce

[Internal Bridge Infill Bug.zip](https://github.com/supermerill/SuperSlicer/files/8102616/Internal.Bridge.Infill.Bug.zip)

**_PS_**: _The instructions on how to export project files are obsolete._ When going to file, there is no "Export Project As" option

### Version

2.3.57

### Operating system

Windows 10

supermerill, 2022-02-19:
Seems like it occurs only when bridge flow ratio is < 64%

supermerill, 2022-02-22:
I investigated and the stl is very bad.
I repaired it with meshmixer, and it's so much better:
[repaired_frame_outer.zip](https://github.com/supermerill/SuperSlicer/files/8117868/repaired_frame_outer.zip)

Sirnut, 2022-02-25:
Is there a way to have superslicer inspect meshes to see if they need repair and then repair them from the app?

supermerill, 2022-03-01:
Prusa already implemented things like that, but it's hard to avoid false positives if your detector is too harsh.

m-sundman, 2022-03-14:
[image]
Is this the problem you're seeing, or is this a different problem?

Sirnut, 2022-10-11:
That looks like the same issue