supermerill/SuperSlicer #2422: Weird layer shifting, but not in Cura ?

pjstam, 2022-02-19:
Weird layer shifting, but not in Cura ?

**Bug description**
I got this box stl file and when slicing an printing it I get a layer shift at the same height. When I slice it in Cura it prints fine. I'm using the default .20 profile from SuperSlicer. I attached the files and it looks that the layer shifts at that litte bevel in the top box. So what I did was cut the top line off and also slicing it with SuperSlicer and Cura. Cura prints fine but SuperSlicer did not (again). I loaded boath gcode fiels generated by SS and Cura into a gcode viewer and they look the same. Anybody know where to start from now?

**To Reproduce**
Steps to reproduce the behavior:
1. Load the stl and slice with default profile...

**Screenshots**
[image]

[image]

[image]

[image]

**Additional context**
Using the latest SS version 2.3.57.11.

supermerill, 2022-02-19:
Your printer probably hit something and layer shift. Try using the z-hop, or watch closely when printing to see what's it's hitting.

pjstam, 2022-02-20:
> Your printer probably hit something and layer shift. Try using the z-hop, or watch closely when printing to see what's it's hitting.

pjstam, 2022-03-04:
Well it looked like something was hitting the walls, played around with some z-hop settings which solved it. Thank you.

closed 2022-03-04 by pjstam