supermerill/SuperSlicer #2423: [2.4 Beta] Using settings from 2.3 SuperSlicer (copying or importing config bundle) causes crash at startup

scottmudge, 2022-02-19:
### What happened?

Copying the settings folder (using manual directory using command line arguments) from 2.3 SuperSlicer to 2.4 beta causes 2.4 to crash at startup. No diagnostic errors given, just crashes to desktop.

If I delete the settings folder, it launches fine and prompts with the Wizard.

If I close the wizard and import a config bundle exported by version 2.3, it won't crash immediately, but if I restart 2.4 Beta, it crashes again.

This effectively mean there is no lateral upgrade path for 2.4, I would have to manually re-add all of the presets and configs.

### Project file & How to reproduce

Start with clean 2.4 settings directory (all defaults), import config bundle provided below.

[Config_Bundle_Example.zip](https://github.com/supermerill/SuperSlicer/files/8103122/Config_Bundle_Example.zip)

### Version

2.4 Beta

### Operating system

Windows 11

### Printer model

MK3S+

supermerill, 2022-02-20:
Can't replicate on my dev build, so hopefully fixed.

scottmudge, 2022-02-20:
I'll try the nightly and if it still crashes I'll throw it in a debugger.

scottmudge, 2022-02-21:
Confirmed fixed in latest nightly.

closed 2022-02-21 by scottmudge