supermerill/SuperSlicer #2424: Marking modifiers on object in preview

CuredPrusa, 2022-02-20:
### Version
Version: 2.3.57.11
Build: SuperSlicer_2.3.57.11
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

### Behavior
I wonder if it is possible to add some kind of shadow on model in preview, over layers affected by Height Range Modifier (and other modifiers). 

When adding code to vertical ruler, it shows darker object's layer in View Color Print, and that's fine. Although it would be even better to have same layer mark in all types of view. 

_Is this a new feature request?_
YES

supermerill, 2022-02-21:
for reference: #2369

supermerill, 2022-02-21:
I'm not 100% sure I understand well. Can you make some crude drawings about what you ned?

CuredPrusa, 2022-02-21:
just to see where are active modifiers present, useful if there are many instances

[image]