supermerill/SuperSlicer #2427: ghost "modified" flag

CuredPrusa, 2022-02-20:
### Version
Version: 2.3.57.11
Build: SuperSlicer_2.3.57.11
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

### Behavior
Each project has separate settings. 
When I toggle between projects (save one, open another), I get Print Settings flagged as modified. But everything was saved/updated before switching to another project.

[image]

supermerill, 2022-02-21:
can't replicate.
can you post here your problematic projects?

CuredPrusa, 2022-02-21:
I can't reproduce either anymore! :(
But you see there was some bug -blank fields. 
Let's ignore until next occurrence.

supermerill, 2022-02-21:
Blank field can appear when you modify something else than the main settings. Like move an object, modify a modifier, adding a part...
I'll try to sort all these things out for a next 2.4 release.

CuredPrusa, 2022-02-21:
I hope Ctrl+S is really saving everything...

supermerill, 2022-02-21:
yes.
Note: when you modify a setting, press enter or click elsewhere before pressing ctrl+S or the modification is applied before saving.