supermerill/SuperSlicer #2428: Artifacts and poor concentric brim ears

mojamuto, 2022-02-20:
### What happened?

[image]

artifacts and poor concentric brim ears on  SuperSlicer-ubuntu_18.04-gtk3-2.3.58.0.AppImage

### Project file & How to reproduce

[prueba.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8104854/prueba.3mf.zip)

### Version

2.3.58.0

### Operating system

Void Linux

### Printer model

Prusa Bear MK3

supermerill, 2022-02-20:
I missused the Internal resolution. Put it to somethign lower (like 0.02) for now.

(Also You should change the thin_perimeters to 100%, it's a conversion problem of this early beta. Using 1% seems to create artifacts.)

mojamuto, 2022-02-21:
It seems that these parameters fix the problems.
thank you very much