supermerill/SuperSlicer #2429: Neúplný překlad do CS

mosnerrene, 2022-02-20:
### What happened?

Zdravím. Nejdřív vám děkuji za skvělou práci co děláte. 
A rád bych se zeptat jestli bude v budoucnosti kompletní překlad do cs jazika?
Dekuji.

### Project file & How to reproduce

SuperSlicer localization cs

### Version

New

### Operating system

Windows 10

### Printer model

Biqu B1

supermerill, 2022-02-21:
Sadely, i can't speak it, so i can't make it. I can use an automatic translation tool like google traduciton, but i'm not sure of the result...

If you are motivated to do it, i can help you start.

SmokeXXL, 2022-02-21:
I can try it. Can you send me a text files?

mosnerrene, 2022-02-25:
Jakýkoliv posun k překladu bude super.
Velmi děkuji.

supermerill, 2022-03-01:
[cs.zip](https://github.com/supermerill/SuperSlicer/files/8165026/cs.zip)

here it is.
The Slic3r.po contains translations from prusa are can be reused.
The TODO contains the untraslated lines.

[here is a little tutorial](https://github.com/supermerill/SuperSlicer/issues/2227#issuecomment-1013786914) on how to use the comments to help you.

renemosner, 2022-03-23:
Děkuji jen nejsem tak dobrý programátor.  Ale přesto bych vám rád pomohl nešlo by udělat krátká instruktážní video jak postupovat krok po kroku. Děkuji za vaší ochotu.

locki-cz, 2022-04-16:
rad testuju, klidne to nahrej sem :)

renemosner, 2022-04-16:
Tak tady to je zkompilovaný. Dej vědět zda je to takto v poho.
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/8500102/SuperSlicer.zip)

locki-cz, 2022-04-17:
> Tak tady to je zkompilovaný. Dej vědět zda je to takto v poho. [SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/8500102/SuperSlicer.zip)
> 
> Když tě něco praští do očí tak dej vědět. Dík.

**Nelegální postavy** by mělo být asi **Zakázané znaky** :) 

Náhled na plátky a Gcode prohlížeč je taky divné, Náhled a G-code náhled?

Pár věcí tam ještě chybí

mosnerrene, 2022-04-17:
> > Tak tady to je zkompilovaný. Dej vědět zda je to takto v poho. [SuperSlicer. zip](https://github.com/supermerill/SuperSlicer/files/8500102/SuperSlicer.zip)
> > Když tě něco praští do očí tak dej vědět. Dík.
> 
> **Nelegální postavy** by mělo být asi **Zakázané znaky** :)
> 
> Náhled na plátky a Gcode prohlížeč je také divné, Náhled a G-code náhled?
> 
> Pár věcí tam ještě chybí

Určitě opravím. Ono je po předkládané v poeditu a moje angličtina není vubec dobrá. 
Když ješte něco najdeš tak dej vědět. 
 A co ješte chybí? 
Ono se mi toho více nevygenerovalo. 
Ale děkuji za jakýkoliv Poznamky.

locki-cz, 2022-04-17:
Snahu cením, ale je to takové že na tom bude jeětě hodně práce, ale nevzdávej to :) Výběr jazyka, Přijdete o obsah talíře :D, místy je to překládané úplně na slepo :( to chce korekcí a korekcí, doporučuji to projít a jak si nejsi jist tak mrknout do prusasliceru jestli tam není něco podobné...

Pár co jsem našel:

#Nastavení tisku:

##Perimetry
Na první vrstvě - Na první - zkráceno at se vleze
Na poslední vrstvě - Na poslední  - zkráceno at se vleze
Vrchní - Vrchních
Spodní - Spodních
Wall Thickness - Tloušťka stěny

Úhlová cena - Cena úhlu
Cestovní náklady - Cena cesty

## Slicování
Výška základní vrstvny - Výška vrstvy
Slice rozolution - Zozlišení slicování

## Výplň
Úhel Výplnit - Výplně
Ladící žehlení - Ladění žehlení

## Podpěry
advanced - Pokročilé

## Rychlost
Default speed - Základní
Rychlost perimetrů - Perimetry
-- Výplň
-- Podpory
-- Mosty
Rychlost cestování - Rychloposun
Rychlost tenkých výtisků - Výplň tenkých stěn
Support acceleration - Zrychlení podpor

renemosner, 2022-04-17:
> Another few fixes and tips in CS transalate:
> 
> Snahu cením, ale je to takové že na tom bude jeětě hodně práce, ale nevzdávej to :) Výběr jazyka, Přijdete o obsah talíře :D, místy je to překládané úplně na slepo :( to chce korekcí a korekcí, doporučuji to projít a jak si nejsi jist tak mrknout do prusasliceru jestli tam není něco podobné...
> 
> Pár co jsem našel:
> 
> #Nastavení tisku:
> 
> ##Perimetry Na první vrstvě - Na první - zkráceno at se vleze Na poslední vrstvě - Na poslední - zkráceno at se vleze Vrchní - Vrchních Spodní - Spodních Wall Thickness - Tloušťka stěny
> 
> Úhlová cena - Cena úhlu Cestovní náklady - Cena cesty
> ## Slicování
> 
> Výška základní vrstvny - Výška vrstvy Slice rozolution - Zozlišení slicování
> ## Výplň
> 
> Úhel Výplnit - Výplně Ladící žehlení - Ladění žehlení
> ## Podpěry
> 
> advanced - Pokročilé
> ## Rychlost
> 
> Default speed - Základní Rychlost perimetrů - Perimetry -- Výplň -- Podpory -- Mosty Rychlost cestování - Rychloposun Rychlost tenkých výtisků - Výplň tenkých stěn Support acceleration - Zrychlení podpor

Díky za korekce.

prosím tě na jakou verzi si to daval já jen že nějaké slova co jsi psal tak přelozené jsou.
ja to bral z té poslední SuperSlicer_2.3.57.12_win64
Tady posílám upravený soubor.
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/8502454/SuperSlicer.zip)

Tak na toprosím tě koukni.

Popřípadě sem hoď vystřižek s popiskem

locki-cz, 2022-04-17:
A tak to me nenapdalo z 2.4.58.2 a ta není stable ...

renemosner, 2022-04-18:
> A tak se mi nenapdalo z 2.4.58.2 a ta není stabilní ...

Tak jak to vidíš budeme pokračovat?

locki-cz, 2022-04-19:
tak jsem zprovoznil stable verzi a musim rict ze tady to ted vypada mnohem lip! Paradni!

jen tady toto vypada blbe, ale to je drobnost, chtelo by to zkratit :)
[image]

Začínat všude velkým písmenem:
[image]

Povolit následné ironing - Zapnout ironing
Ladění ironing - Ladění

[image]

renemosner, 2022-04-20:
> tak jsem zprovoznil stable verzi a musim rict ze tady to ted vypada mnohem lip! Paradni!
> 
> jen tady toto vypada blbe, ale to je drobnost, chtelo by to zkratit :) [image]
> 
> Začínát všude velkým písmenem: [image]
> 
> Povolit následné žehlení - Zapnout žehlení Ladění žehlení - Ladění
> 
> [image]

Diky za komenty tady posilam posledni upravený.  Jen to skraceni jsem neresil.
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/8524873/SuperSlicer.zip)

locki-cz, 2022-04-20:
Chtelo by to nekoho dalsiho co by nasel mouchy co ja nevidim :)

mosnerrene, 2022-04-24:
> Chtelo by to nekoho dalsiho co by nasel mouchy co ja nevidim :)

Ahoj prosím tě kde seženu tu novou neofiko verzi ke stažení že bych ji skusil překládat? Jen aby byla hotova a nemusel bych ji bildit.

locki-cz, 2022-04-24:
releasy jsou zde: https://github.com/supermerill/SuperSlicer/releases jak se kompiluje preklad netusim :(

renemosner, 2022-04-24:
> releasy jsou zde: https://github.com/supermerill/SuperSlicer/releases jak se kompiluje překlad netusim :( 
A ty máš jaký Kompilace překladu umím ale myslel jsem zkompilovaný superslicer jako máš ty nebo to musím nějak poslepovat?

Vím že jsem otravnej.

locki-cz, 2022-04-24:
Pod tim releasem mas zip soubor ktery staci rozbalit a je to funkcni jen to je fishy edition ktera zatim neni oznacovana za stable, asi to ma porad nejakou mouchy :), cestinu nakopurujes do slozky s reources/cs mam jen to co je zde

renemosner, 2022-04-24:
> Pod tim releasem mas zip soubor ktery staci rozbalit a je to funkcni jen to je fishy edition ktera zatim neni oznacovana za stabilni, asi to ma nejako poradu mouchy :), cestinu nakopurujes do slozky s reources/cs a je to, nic sloziteho.
> 
> [image]

Joooo už to mám. A ty máš ten samý já jestli pak můžu poslat něco na otestování.

locki-cz, 2022-04-24:
jo pouzivam tento, nevim cim se to lisi ze nektere veci co jsou prelozene v te stable verzi se nezobrazuji tady v Pre-release

renemosner, 2022-04-24:
> jo pouzivam tento, nevim cim se to lisi ze nektere veci co jsou prelozene v te stabilni verzi se nezobrazuji tady v Pre-release

Ahoj tady je Nějaký překlad koukni na to prosím.
Některé věci přeložit zatím nejdou python mi nevygeneruje k tomu cestu. 

[image]
[image]

[Superslicer v2.4.58.2.zip](https://github.com/supermerill/SuperSlicer/files/8550411/Superslicer.v2.4.58.2.zip)

Dej pak vědět zda je to v poho.

locki-cz, 2022-04-24:
na tu nestable verzi kasli, neni dopecena :)