supermerill/SuperSlicer #2431: NEW FEATURE REQUEST - Top/bottom generator on-off per layer

CuredPrusa, 2022-02-21:
### Version
Version: 2.3.57.11
Build: SuperSlicer_2.3.57.11
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

### Description
Is there something like "top surface is only the highest layer /bottom the lowest layer"?
It would be useful to have it.
I have to make some holes inside the model, where I manually put some metal components. I wouldn't like slicer to treat it like surface with bottom and top. Better structual consistecy would be achieved if inserts are placed directly on dense integrated infill, and covered by the same (rectlinear). I'm aware this requirement could be rare, but if it's not difficult to put a tick mark for "top surface is only the highest layer /bottom the lowest layer", I'd be more than grateful. 

On the long run, the best option would be to have available Modifier override -"ignore top/bottom"  
(presently available "Top/bottom solid layer" doesn't help in Modifier)

To achieve desired quality now, I have to sacrifice real top and bottom, by disabling.

supermerill, 2022-02-21:
with modifiers:
[image]
[no_solid_inside.zip](https://github.com/supermerill/SuperSlicer/files/8109591/no_solid_inside.zip)

CuredPrusa, 2022-02-21:
That looks promising! 
Thanks, I'll try!

CuredPrusa, 2022-02-21:
Guys, sorry for unintentional spamming.
I guess I’m just too tired…
I guess I’ve clicked something wrong and didn’t get the proper effect from "Top/bottom solid layer" .
That’s why I wrote:  (presently available "Top/bottom solid layer" doesn't help in Modifier)
It DOES work, exactly as expected, exactly what I need! 

SuperSlicer amazes me day after day.
I curse all those non-refundable months spent with Cura….

closed 2022-02-21 by CuredPrusa