supermerill/SuperSlicer #2433: Export a scaled stl

eddy-99, 2022-02-21:
### Version
_Version of SuperSlicer used goes here (help->about)_
2.4.58.0
### Operating system type + version
_What OS are you using, and state any version #s_
Windows 11
Import stl, scale to a new size and export back to same stl, file does not overwrite the original file with new size

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

neophyl, 2022-02-22:
Also in PS https://github.com/prusa3d/PrusaSlicer/issues/7464

supermerill, 2022-02-22:
I tested it with PS2.4 and 2.3.3, and it works with these.

liftbag, 2022-02-24:
> I tested it with PS2.4 and 2.3.1, and it works with these. but there is the same bug in 2.4.1, maybe from the " 'Export to STL' fix " So I'll it rot, hopefully resolved by the final release of PS 2.4.1

Solved in PrusaSlicer 2.4.1 Beta3.