supermerill/SuperSlicer #2434: One perimeter on bottom layer doesn't work anymore

LazaroFilm, 2022-02-21:
### What happened?

I like to create a layer modifier to make the first layer single walled like the top layer. It used to work well, but on the last version, it keeps the walls and widens the infill, making them overlay
[image]
.

### Project file & How to reproduce

Any file, create a height range modifier for the first layer then chose only one perimeter.

### Version

Final 2.3

### Operating system

MacOS 12.2.1

### Printer model

HEVO

supermerill, 2022-02-21:
can't reproduce
[image]

your **project file** ?

supermerill, 2022-02-21:
Can't reproduce
[image]
Your **project file** please

LazaroFilm, 2022-02-23:
Ok, looked deeper. It somehow is linked to the settings of my print profile. I still don't know which combination of settings created that issue, but switching to another print profile solved the issue.

supermerill, 2022-02-23:
You don't want this combination of setting to be fixed?

LazaroFilm, 2022-02-25:
I would, but I don't even know what the combination is. I think it has to do with reusing an old Config with the new version and some settings getting mixed up. Also, I just realized that you added a single perimeter on first layer option which works better and is faster to setup than my intricate layer modifier (thank you for that)

LazaroFilm, 2022-06-12:
After cleaning all old configs and reinstalling, everything works.

closed 2022-06-12 by LazaroFilm