supermerill/SuperSlicer #2435: Seam position "rear" is very messy

alexvonsiebenthal, 2022-02-21:
### What happened?

I struggle with SS to get clean seams. I really miss the "Sharpest Corner" option from Cura and I can't get it to work with the cost-based option either. It's still all over the place. The best - but still not good - way I found is to rotate the model and set the seam position to "rear". However, even then it's messy. Why can't it just all be aligned like it is in the middle of the print?
[image]

### Project file & How to reproduce

[SeamRearIssue.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8112000/SeamRearIssue.3mf.zip)

### Version

2.3.57.9

### Operating system

macOS 12.1

### Printer model

voron 2.4

supermerill, 2022-02-21:
[SeamRearIssue.zip](https://github.com/supermerill/SuperSlicer/files/8112026/SeamRearIssue.zip)

if you want a precise control, use seam spheres (it's in the contextual menu)

I'll tag this issue to look at the rear.

>  really miss the "Sharpest Corner" option from Cura

There is no real sharp corner in your model, the external curve in each one is seen as  very gentle.

alexvonsiebenthal, 2022-02-21:
Holy. Never seen that sphere before but it is able to produce some really nice seams indeed. Not the easiest UX though. Thank you.

As some food for thought: Manual seem painting would also benefit from options like being able to draw straight lines (maybe using Shift key or similar). I'm not good at seam painting with the mouse cursor.

supermerill, 2022-02-21:
imo, seam painting is good for when you draw them on a very sharp corner and set the seam to corner-weight, so it will go in the corner. Other than that...

fred099, 2022-02-25:
I have the same similar issues and noticed it is caused by the detraction after a short travel at high acceleration. This is from the inside with 15k acceleration.

I'm looking forward to trying the new acceleration features when the MacOS arm nightly build is working. I think it addresses this problem.
[image]
[image]