supermerill/SuperSlicer #2438: Question about shifting modifiers and custom gcode

CuredPrusa, 2022-02-22:
### Version
Version: 2.3.57.11
Build: SuperSlicer_2.3.57.11
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

### Question
I have to add 8mm on the base of my model. That will lift the whole model up by 8mm.
Does that mean that I have to manually rewrite 23 custom gcodes and 12 Modifier ranges?
Or is there some trick to shift all at once?

CuredPrusa, 2022-02-22:
Thanks!
I'll try what you proposed. 
 Since we're at "hacking", I have one problem with print initialization.
Can you tell how to change gcode generator from this what we have now:  

**M140 ; set bed temperature**
**M104 ; set temperature**
;TYPE:Custom
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; Filament gcode
**M109 ; set temperature and wait for it to be reached**
**M190 ; set bed temperature and wait for it to be reached**
;LAYER_CHANGE

to this:

**M104 ; set temperature**
**M190 ; set bed temperature and wait for it to be reached**
;TYPE:Custom
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; Filament gcode
;LAYER_CHANGE

Until I figure that out, I have to change each gcode file manually. 
I need it for my BLTouch to work properly (bed heated, hence warped).

neophyl, 2022-02-22:
Just add the lines you want to the start gcode section in Printer Settings>Custom gcode section.  That's not hacking, it's just properly setting up your printer.

If you post a zipped up saved project then we can see your current start gcode (as well as the rest of the settings) and tell you exactly what to put where or even modify your project and upload a new version so you can see any changes.

CuredPrusa, 2022-02-22:
Take a look once again. Lines that bother me are BEFORE and AFTER custom gcode. Those can't be influenced by custom gcode or filament code.
Yes, I'm using custom gcode, but I deleted it from example because it's irrelevant.

neophyl, 2022-02-22:
I took a look at all your other posts trying to fins any instance of a saved project.  I only managed to find one from you in 12 issues you opened.  Called Testflow_BE.  You used a 0.8mm nozzle etc.  Is that still your normal printer ?  The start code you deleted from your examples is not irrelevant though.  You just don't have all the required information about how it works.

Normally you should always include the heating lines in start gcode.  Take a look at any of the predefined profiles for the major printers like a Prusa MK3 or a creality as an example.
As a safety measure if you do not manually add the lines for setting the extruder and bed heater to the custom start gcode then Super Slicer will automatically add them.  As it cant print cold.   Unless the tickbox "Only custom start gcode" is ticked at the top of that section in which case whatever happens is on you.  Your saved project file does not have that ticked so its adding them for you.  However its adding them at a certain point in the logic.

If you manually add them then YOU get to control where and when they are inserted.  
Your current start gcode section from that project appears to be the following
G28 ; home all axes
G29 ;
G1 Z5 F5000 ; lift nozzle
G28 ; home all axes
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 X14.1 Y20 Z0.28 F5000.0 ;Move to start position2
G1 X14.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line2
G1 X14.4 Y200.0 Z0.28 F5000.0 ;Move to side a little2
G1 X14.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line2
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
M300 S440 P200

Note there no temperature commands in there. If you add the following lines as an example 

G28 ; home all axes
G29 ;
G1 Z5 F5000 ; lift nozzle
G28 ; home all axes 
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
M140 S[first_layer_bed_temperature] ; set bed final temp
M104 S[first_layer_temperature] ; set extruder final temp
M109 S[first_layer_temperature] ; wait for extruder final temp
M190 S[first_layer_bed_temperature] ; wait for bed final temp
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 X14.1 Y20 Z0.28 F5000.0 ;Move to start position2
G1 X14.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line2
G1 X14.4 Y200.0 Z0.28 F5000.0 ;Move to side a little2
G1 X14.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line2
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
M300 S440 P200

Then it would move your z axis up, preheat and wait for the first layer extruder and bed temperatures to be reached and only then move to your start position.  

You can do other stuff in there too, like setting the extruder to a no ooze temp of around 160 until the bed has heated up and only then finishing heating the extruder all the way.  That gets your bed up to temp without filament leaking out waiting with a fully hot extruder.  That's the whole point of the custom start gcode sections, to make it behave as you want it to.

CuredPrusa, 2022-02-22:
OMG neophyle, you’ve just become  my new best friend!

I was totally unaware that software is taking care of covering undefined
mandatory instructions from custom gcode. I also didn’t know how to call a
set point from project [ ]. I’m very grateful for your help and effort you
took to explain. Thank you so much!